OPEnSLab-OSU/SSLClient #3: Add support for STM32 Ethernet

gdsports, 2019-10-18:
First attempt at supporting STM32 Nucleo F767ZI with built-in Ethernet. Cortex M7 @ 216 MHz, 2 Mbyte Flash. Two of three examples working.

https://github.com/gdsports/SSLClient/tree/stm32ethernet

https://github.com/stm32duino/STM32Ethernet/issues/14

prototypicalpro, 2019-10-18:
Is your code ready for a pull request? I'd be happy to merge it and add official support if so.

gdsports, 2019-10-18:
Not ready yet because the EthernetMultiHTTPS example is failing for STM32. I hope to look at this more next week.

gdsports, 2019-10-21:
I pushed the PR. The fix https://github.com/stm32duino/STM32Ethernet/pull/28 for EthernetMultiHTTPS is in STM32Ethernet so when the fix, merged all TLS examples work.

fpistm, 2019-11-10:
Hi @gdsports, @prototypicalpro,
all has been merged and are available in release 1.0.5.

prototypicalpro, 2020-03-09:
Fixed with https://github.com/OPEnSLab-OSU/SSLClient/commit/8cbf028b986429e78595b5289bb3f7c50ce1a3cb.

closed 2020-03-09 by prototypicalpro

joaquinbvw, 2021-01-12:
Hi. First of all, thank you for the STM32 support, I think STM32 is gaining a lot of fans and is great to exploit all their benefits. Regarding the example EthernetHTTPSstm32.ino I had problems running it with static IP address. Can't say what is the issue, I tried to change the trust anchors but it seems that they are fine as they are. It says "Error writing to m_client" but I don't know what that means. Below I'm putting the serial terminal output, with DHCP it works fine btw, maybe is on the Ethernet library side?

13:39:22.037 -> Initialize Ethernet with DHCP:
13:39:26.074 -> 192.168.8.110
13:39:28.042 -> connecting to www.arduino.cc...
13:39:37.878 -> (SSLClient)(SSL_ERROR)(m_update_engine): Error writing to m_client
13:39:37.878 -> (SSLClient)(SSL_ERROR)(m_update_engine): 0
13:39:37.878 -> (SSLClient)(SSL_ERROR)(connected): Not connected because write error is set
13:39:37.878 -> (SSLClient)(SSL_ERROR)(m_print_ssl_error): SSL_CLIENT_WRITE_FAIL
13:39:37.878 -> (SSLClient)(SSL_WARN)(m_run_until): Terminating with write error: 
13:39:37.878 -> (SSLClient)(SSL_WARN)(m_run_until): 4
13:39:37.878 -> (SSLClient)(SSL_ERROR)(m_start_ssl): Failed to initlalize the SSL layer
13:39:37.878 -> (SSLClient)(SSL_ERROR)(m_print_br_error): Unknown error code: 0
13:39:37.878 -> connection failed
13:39:37.878 -> (SSLClient)(SSL_ERROR)(available): Cannot operate if the write error is not reset: 
13:39:37.878 -> (SSLClient)(SSL_ERROR)(m_print_ssl_error): SSL_CLIENT_WRITE_FAIL
13:39:37.878 -> (SSLClient)(SSL_ERROR)(connected): Not connected because write error is set
13:39:37.878 -> (SSLClient)(SSL_ERROR)(m_print_ssl_error): SSL_CLIENT_WRITE_FAIL
13:39:37.878 -> 
13:39:37.878 -> disconnecting.
13:39:37.878 -> (SSLClient)(SSL_ERROR)(connected): Not connected because write error is set
13:39:37.878 -> (SSLClient)(SSL_ERROR)(m_print_ssl_error): SSL_CLIENT_WRITE_FAIL
13:39:37.878 -> Received 0 bytes in 0.0001, rate = 0.00 kbytes/second