OPEnSLab-OSU/SSLClient #11: Can't find Client.h

saurin-tech, 2020-05-24:
Hi,

I'm trying to utilize the SSLClient with the AZ3166WiFiClient for the MXChip.  But When I try to compile the code I keep getting an error stating:
`fatal error: Client.h: No such file or directory`

In my c_cpp_properties.json file I've listed `"/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino/**"` to the includePath which has the Client.h

Is there something wrong that I'm doing or can this library be even utilized for AZ3166 MXCHIP?

Thanks!

prototypicalpro, 2020-05-24:
I'm assuming you're using [this Arduino core](https://github.com/microsoft/devkit-sdk/tree/master/AZ3166/src/cores/arduino) by microsoft? If so, it looks like they forgot just to include Client.h as part of the core. I opened https://github.com/microsoft/devkit-sdk/issues/1067 to address this, but in the meantime you can add [this version of Client.h](https://github.com/arduino/ArduinoCore-samd/blob/master/cores/arduino/Client.h) to you project and that should fix it.

saurin-tech, 2020-05-24:
@prototypicalpro 
Yes, I'm using the Arduino core by Microsoft.  I've tried copying and pasting the Client.h that you've mentioned in the above conversation.  But I'm still running into issues due to the conversion of Microsoft WiFiClient from `Az3166WiFiClient.h` to Client that is being utilized by SSLClient.

`app:216:61: error: no matching function for call to 'SSLClient::SSLClient(WiFiClient&, const br_x509_trust_anchor [2], size_t, const int&)'
 SSLClient client(base_client, TAs, (size_t)TAs_NUM, rand_pin);
                                                             ^
In file included from /Users/saurin/AzureIOT/MXChip-temp/app.ino:9:0:
/Users/saurin/Documents/Arduino/libraries/SSLClient/src/SSLClient.h:94:14: note: candidate: SSLClient::SSLClient(Client&, const br_x509_trust_anchor*, size_t, int, size_t, SSLClient::DebugLevel)
     explicit SSLClient( Client& client, 
              ^
/Users/saurin/Documents/Arduino/libraries/SSLClient/src/SSLClient.h:94:14: note:   no known conversion for argument 1 from 'WiFiClient' to 'Client&'
/Users/saurin/Documents/Arduino/libraries/SSLClient/src/SSLClient.h:34:7: note: candidate: SSLClient::SSLClient(const SSLClient&)
 class SSLClient : public Client {`

Thanks for prompt reply!

prototypicalpro, 2020-05-24:
The strange thing is that `Az3166WiFiClient` inherits from `Print`, but judging by the interface it looks like it was designed to inherit from `Client` and was just never fixed. I added a comment to https://github.com/microsoft/devkit-sdk/issues/1067 about this, however what you might try is editing your copy of [AZ3166WiFiClient.h](https://github.com/microsoft/devkit-sdk/blob/master/AZ3166/src/libraries/WiFi/src/AZ3166WiFiClient.h#L23) to replace this:
```C++
class WiFiClient : public Print
```
with this:
```C++
#include "Client.h"
class WiFiClient : public Client
```

This may or may not compile, but it's certainly worth a shot while we wait for the devkit-sdk to be fixed.

saurin-tech, 2020-05-24:
After applying the changes you suggested it pointed me to another error which is the discrepancy between the return type of the connected method.

`In file included from /Users/saurin/Library/Arduino15/packages/AZ3166/hardware/stm32f4/1.6.5/libraries/WiFi/src/AZ3166WiFi.h:23:0,
                 from /Users/saurin/AzureIOT/MXChip-temp/app.ino:1:
/Users/saurin/Library/Arduino15/packages/AZ3166/hardware/stm32f4/1.6.5/libraries/WiFi/src/AZ3166WiFiClient.h:43:15: error: conflicting return type specified for 'virtual int WiFiClient::connected()'
   virtual int connected();
               ^
In file included from /Users/saurin/Library/Arduino15/packages/AZ3166/hardware/stm32f4/1.6.5/libraries/WiFi/src/AZ3166WiFiClient.h:23:0,
                 from /Users/saurin/Library/Arduino15/packages/AZ3166/hardware/stm32f4/1.6.5/libraries/WiFi/src/AZ3166WiFi.h:23,
                 from /Users/saurin/AzureIOT/MXChip-temp/app.ino:1:
/Users/saurin/Documents/Arduino/libraries/SSLClient/src/Client.h:36:19: error:   overriding 'virtual uint8_t Client::connected()'
   virtual uint8_t connected() = 0;`

I changed the return type on the Microsoft side to uint8_t that resulted in another error. So I reverted my changes and changed the SSLClient to int, that resulted in the same error that I'm posting below.  

`In file included from /Users/saurin/Library/Arduino15/packages/AZ3166/hardware/stm32f4/1.6.5/cores/arduino/Print.h:26:0,
                 from /Users/saurin/Documents/Arduino/libraries/SSLClient/src/Client.h:19,
                 from /Users/saurin/Documents/Arduino/libraries/SSLClient/src/SSLClient.h:21,
                 from /Users/saurin/Documents/Arduino/libraries/SSLClient/src/SSLClient.cpp:21:
/Users/saurin/Library/Arduino15/packages/AZ3166/hardware/stm32f4/1.6.5/system/mbed-os/platform/Callback.h: In member function 'void mbed::Callback<R()>::generate(const F&)':
/Users/saurin/Library/Arduino15/packages/AZ3166/hardware/stm32f4/1.6.5/cores/arduino/WString.h:38:29: error: expected type-specifier before '(' token
 #define FPSTR(pstr_pointer) (reinterpret_cast<const __FlashStringHelper *>(pstr_pointer))
                             ^
/Users/saurin/Library/Arduino15/packages/AZ3166/hardware/stm32f4/1.6.5/cores/arduino/WString.h:39:28: note: in expansion of macro 'FPSTR'
 #define F(string_literal) (FPSTR(PSTR(string_literal)))
                            ^
/Users/saurin/Library/Arduino15/packages/AZ3166/hardware/stm32f4/1.6.5/system/mbed-os/platform/Callback.h:608:20: note: in expansion of macro 'F'
         new (this) F(f);
                    ^
/Users/saurin/Library/Arduino15/packages/AZ3166/hardware/stm32f4/1.6.5/cores/arduino/WString.h:38:29: error: expected ')' before '(' token
 #define FPSTR(pstr_pointer) (reinterpret_cast<const __FlashStringHelper *>(pstr_pointer))
                             ^
/Users/saurin/Library/Arduino15/packages/AZ3166/hardware/stm32f4/1.6.5/cores/arduino/WString.h:39:28: note: in expansion of macro 'FPSTR'
 #define F(string_literal) (FPSTR(PSTR(string_literal)))
                            ^
/Users/saurin/Library/Arduino15/packages/AZ3166/hardware/stm32f4/1.6.5/system/mbed-os/platform/Callback.h:608:20: note: in expansion of macro 'F'
         new (this) F(f);
                    ^
/Users/saurin/Library/Arduino15/packages/AZ3166/hardware/stm32f4/1.6.5/system/mbed-os/platform/Callback.h: In static member function 'static void mbed::Callback<R()>::function_move(void*, const void*)':
/Users/saurin/Library/Arduino15/packages/AZ3166/hardware/stm32f4/1.6.5/cores/arduino/WString.h:38:29: error: expected type-specifier before '(' token
 #define FPSTR(pstr_pointer) (reinterpret_cast<const __FlashStringHelper *>(pstr_pointer))
                             ^
/Users/saurin/Library/Arduino15/packages/AZ3166/hardware/stm32f4/1.6.5/cores/arduino/WString.h:39:28: note: in expansion of macro 'FPSTR'
 #define F(string_literal) (FPSTR(PSTR(string_literal)))
                            ^
/Users/saurin/Library/Arduino15/packages/AZ3166/hardware/stm32f4/1.6.5/system/mbed-os/platform/Callback.h:620:17: note: in expansion of macro 'F'
         new (d) F(*(F*)p);
                 ^
/Users/saurin/Library/Arduino15/packages/AZ3166/hardware/stm32f4/1.6.5/cores/arduino/WString.h:38:29: error: expected ')' before '(' token
 #define FPSTR(pstr_pointer) (reinterpret_cast<const __FlashStringHelper *>(pstr_pointer))
                             ^
/Users/saurin/Library/Arduino15/packages/AZ3166/hardware/stm32f4/1.6.5/cores/arduino/WString.h:39:28: note: in expansion of macro 'FPSTR'
 #define F(string_literal) (FPSTR(PSTR(string_literal)))
                            ^
/Users/saurin/Library/Arduino15/packages/AZ3166/hardware/stm32f4/1.6.5/system/mbed-os/platform/Callback.h:620:17: note: in expansion of macro 'F'
         new (d) F(*(F*)p);
                 ^
/Users/saurin/Library/Arduino15/packages/AZ3166/hardware/stm32f4/1.6.5/system/mbed-os/platform/Callback.h: In static member function 'static void mbed::Callback<R()>::function_dtor(void*)':
/Users/saurin/Library/Arduino15/packages/AZ3166/hardware/stm32f4/1.6.5/cores/arduino/WString.h:39:27: error: expected identifier before '(' token
 #define F(string_literal) (FPSTR(PSTR(string_literal)))
                           ^
/Users/saurin/Library/Arduino15/packages/AZ3166/hardware/stm32f4/1.6.5/system/mbed-os/platform/Callback.h:625:19: note: in expansion of macro 'F'
         ((F*)p)->~F();
                   ^
[... cut ...]
                                                                            ^
/Users/saurin/Library/Arduino15/packages/AZ3166/hardware/stm32f4/1.6.5/cores/arduino/WString.h:39:34: note: in expansion of macro 'PSTR'
 #define F(string_literal) (FPSTR(PSTR(string_literal)))
                                  ^
/Users/saurin/Library/Arduino15/packages/AZ3166/hardware/stm32f4/1.6.5/system/mbed-os/events/Event.h:2204:25: note: in expansion of macro 'F'
         ((F*)(e + 1))->~F();
                         ^`

prototypicalpro, 2020-05-25:
I'm guessing that error is because changing the `SSLClient::connected` return type causes it to not implement the `Print` interface properly, though the error you sent is nearly impossible to read. I think the best course of action from here is to wait for the devkit-sdk team to respond to https://github.com/microsoft/devkit-sdk/issues/1067, which should address this issue.

saurin-tech, 2020-05-25:
Sounds good.  Thanks for your help!

prototypicalpro, 2021-03-25:
I'm going to go ahead and close this issue since it's not (technically) an issue with SSLClient, hopefully microsoft responds to https://github.com/microsoft/devkit-sdk/issues/1067 eventually :(.

closed 2021-03-25 by prototypicalpro