OPEnSLab-OSU/SSLClient #13: Calls to stop before connected cause lockups.

bleckers, 2020-06-14:
**Describe the bug**
Calling sslClient.stop() when !sslClient.connected() causes buffer overflow. 

**To Reproduce**
SSLClient sslClient(httpsEthClient, TAs_HTTPS, (size_t)TAs_NUM_HTTPS, RANDOM_DATA_PIN);
sslClient.stop();

**Expected behavior**
Calls to stop should exit gracefully or ignore the request if already exited.

**Context (please complete the following information):**
 - Device Type - ESP32/Teensy4
 - Arduino Core Version - Teensyduino 1.52
 - Relevant Library Versions - PubSubClient 
 - SSLClient Version - v1.6.6

**Additional context**
Removing jump_handshake in br_ssl_engine_close prevents this (not a fix).

prototypicalpro, 2020-06-14:
Is the behavior a crash or a freeze? In my experience BearSSL attempts to close the connection by sending a [`close_notify`](https://security.stackexchange.com/questions/82028/ssl-tls-is-a-server-always-required-to-respond-to-a-close-notify) (which is usually ignored) and then times out waiting for a response.

prototypicalpro, 2020-06-14:
I think https://github.com/OPEnSLab-OSU/SSLClient/commit/4707ea4b48aff3474f4b15fc1e754326cb5c22d3 should fix the `close_notify` issue, would you give master latest a shot?

bleckers, 2020-06-15:
The behaviour is a crash rather than a freeze. The new commit fixes this with a fresh stop before connect, after a client has connected and disconnected AND if the client is connected.

closed 2020-06-15 by bleckers

bleckers, 2020-06-15:
Oh just to add, if you have network connectivity but no connection to the server and are using the Arduino Ethernet 2.0 library, you will need the following change in the EthernetClient.cpp for this commit to fix the issues (as calls to get_arduino_client().flush(); in this configuration will hang indefinitely). This is also true for calls to flush in the m_update_engine of SSLClient, and considering the lack of development effort in Ethernet, it might have to be a recommendation in SSLClient (particularly if using PubSubClient because of the regular pings which are susceptible to minor lapses in network connectivity).

```
void EthernetClient::flush()
{
	unsigned long start = millis();
	while (sockindex < MAX_SOCK_NUM) {
		if (millis() - start > _timeout) 
		{
			stop();
			return;
		}
		uint8_t stat = Ethernet.socketStatus(sockindex);
		if (stat != SnSR::ESTABLISHED && stat != SnSR::CLOSE_WAIT) return;
		if (Ethernet.socketSendAvailable(sockindex) >= W5100.SSIZE) return;
	}
}
```

prototypicalpro, 2020-06-15:
Hmmm yeah, I don't think I have a good way of enforcing a timeout on `Client::flush` without modifying the client implementation. I will add a note in the README about this, and patch [EthernetLarge](https://github.com/OPEnSLab-OSU/EthernetLarge) as soon as your pull request is merged.

commit f40b9c47 (2020-06-15): Merge pull request #14 from OPEnSLab-OSU/develop

Fix #13