OPEnSLab-OSU/SSLClient #15: Basic HTTPS get on esp32 using WiFiClient not compiling

MrMoine, 2020-08-28:
I just downloaded the library and tried to use it on my ESP32, using the Client `WiFiClient` but it won't compile because it says that `SSLClient` is abstract. I tried to look for this issue on internet but I couldn't find any forum talking about it. So here is my code (I'm using the Arduino IDE btw)
```cpp
#include <WiFiClient.h>
#include <SSLClient.h>
#include "cert.h"

const char *ssid = "cathjphi";
const char *password = "leschatons";
const char *server = "api.wit.ai";
const char *endpoint = "/message?q=%C3%A9teins+la+lumi%C3%A8re";

const int port = 443;

WiFiClient baseClient;
const int rand_pin = 33;
SSLClient client(baseClient, TAs, (size_t)TAs_NUM, rand_pin);

void setup()
{
	WiFi.begin(ssid, password);

	Serial.println("Connection au WiFi");
	while (WiFi.status() != WL_CONNECTED)
	{
		delay(500);
		Serial.print(".");
	}
	Serial.print("Connecté au WiFi avec l'ip ");
	Serial.println(WiFi.localIp());

	if (client.connect(server, 443))
	{
		client.println(String("GET https://") + server + endpoint + " HTTP/1.1");
		client.println("User-Agent: SSLClientOverEthernet");
		client.print("Host: ");
		client.println(server_host);
		client.println("Authorization: Bearer YSPRKSUB6I3NEQ6U6WTH5URHRZXSEZ45");
		client.println("Connection: close");
		client.println();
	}
	else
	{
		Serial.println("connection failed");
	}
}

void loop()
{
	int len = client.available();
	if (len > 0)
	{
		byte buffer[80];
		if (len > 80)
			len = 80;
		client.read(buffer, len);
		if (printWebData)
		{
			Serial.write(buffer, len); // show in the serial monitor (slows some boards)
		}
		byteCount = byteCount + len;
	}
}
```
And here is the whole compiler error :
```SSLClient_test:14:11: error: cannot declare variable 'client' to be of abstract type 'SSLClient'
 SSLClient client(baseClient, TAs, (size_t)TAs_NUM, rand_pin);
           ^
In file included from C:\Users\Antonin\Documents\Arduino\google home maison\SSLClient_test\SSLClient_test.ino:2:0:
C:\Users\Antonin\Documents\Arduino\libraries\SSLClient-master\src/SSLClient.h:34:7: note:   because the following virtual functions are pure within 'SSLClient':
 class SSLClient : public Client {
       ^
In file included from C:\Users\Antonin\Documents\ArduinoData\packages\esp32\hardware\esp32\1.0.2\cores\esp32/Arduino.h:157:0,
                 from sketch\SSLClient_test.ino.cpp:1:
C:\Users\Antonin\Documents\ArduinoData\packages\esp32\hardware\esp32\1.0.2\cores\esp32/Client.h:31:17: note: 	virtual int Client::connect(IPAddress, uint16_t, int)
     virtual int connect(IPAddress ip, uint16_t port, int timeout) =0;
                 ^
C:\Users\Antonin\Documents\ArduinoData\packages\esp32\hardware\esp32\1.0.2\cores\esp32/Client.h:32:17: note: 	virtual int Client::connect(const char*, uint16_t, int)
     virtual int connect(const char *host, uint16_t port, int timeout) =0;
                 ^
C:\Users\Antonin\Documents\Arduino\google home maison\SSLClient_test\SSLClient_test.ino: In function 'void setup()':
SSLClient_test:18:2: error: 'WiFi' was not declared in this scope
  WiFi.begin(ssid, password);
  ^
SSLClient_test:21:26: error: 'WL_CONNECTED' was not declared in this scope
  while (WiFi.status() != WL_CONNECTED)
                          ^
SSLClient_test:34:18: error: 'server_host' was not declared in this scope
   client.println(server_host);
                  ^
C:\Users\Antonin\Documents\Arduino\google home maison\SSLClient_test\SSLClient_test.ino: In function 'void loop()':
SSLClient_test:54:7: error: 'printWebData' was not declared in this scope
   if (printWebData)
       ^
SSLClient_test:58:3: error: 'byteCount' was not declared in this scope
   byteCount = byteCount + len;
   ^
Plusieurs bibliothèque trouvées pour "WiFiClient.h"
Utilisé : C:\Users\Antonin\Documents\ArduinoData\packages\esp32\hardware\esp32\1.0.2\libraries\WiFi
Non utilisé : C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.42.0_x86__mdqgnx93n4wtt\libraries\WiFi
exit status 1
cannot declare variable 'client' to be of abstract type 'SSLClient'
```
I hope there is enough information and that I can get help soon enough.
Thanks ^^

prototypicalpro, 2020-08-28:
Hey there! Have you tried adding `#include <WiFi.h>` to the top of your sketch?

MrMoine, 2020-08-28:
No, I just tried it but it still don't compile, thanks for the quick answer btw 😄

prototypicalpro, 2020-08-29:
No problem! It looks like the cause of the compiler errors is your version of the ESP32 core (`v1.0.2`). The ESP32 core added some [functions to the Client interface](https://github.com/espressif/arduino-esp32/blob/6bf104874a453596ac4b2d71feb9917ddaea53fd/cores/esp32/Client.h#L31-L32) in `v1.0.2`, causing a lot of libraries to break and [ruffling some feathers](https://github.com/espressif/arduino-esp32/commit/9a9ff622166149da30dd13a81485184617c56387#commitcomment-33292990). This change was [reverted](https://github.com/espressif/arduino-esp32/commit/f5cacfee1a2dfaf693afa8d610e8f2c92b74f375#diff-0336fe0dc43715cecf17987314b9070b) in `v1.0.3` due to the aforementioned feather ruffling. If you upgrade your ESP32 core to `v1.0.3` or above (or downgrade below `v1.0.2` if you like) the sketch should compile. Let me know if that doesn't work and I'd be happy to investigate further.

Also: it looks like you posted credentials in your example. As these issues are publicly viewable to anyone, you will want to invalidate those credentials so no one can use them maliciously.

closed 2020-08-30 by prototypicalpro

prototypicalpro, 2020-09-01:
I noticed you edited the credentials out of your comment. For future reference, GitHub keeps an [edit history](https://docs.github.com/en/github/building-a-strong-community/tracking-changes-in-a-comment) on issues which is also viewable by anyone. The edit history can be scrubbed by removing all revisions with the sensitive data in the `edited` menu. I went ahead and scrubbed your comment, so your credentials should be fully off GitHub now 😃. Since your credentials have been up for a few days, however, I _highly_ recommend you regenerate them if you haven't already.