OPEnSLab-OSU/SSLClient #16: Is there a difference between this client and python ssl module | esp32 https chunked encoding

MrMoine, 2020-09-18:
Hello, I want to use this client to stream audio to the `wit.ai` speech recognition api using HTTP chunked encoding.
I've been having problem since quite sometime now so in order to debug in a simpler way I've written a similar program in python.
So I used the `socket` and the `ssl` module of python assuming they'd be of the same level as this client. So here is my python code that I managed to make working : 
```python
import socket
import ssl
import time

def requestLog(ssock, string):
	print(string, end="")
	ssock.sendall(bytes(string, "ascii"))

hostname = 'api.wit.ai'
endpoint = "/speech"
context = ssl.create_default_context()
BIT_DEPTH = 8
RATE = 50000
f = open("./main/test.raw", "rb")#i'm reading a file to replicate the esp32 microphone (the data in the file come from the esp32
buffers = []
while True:
	buffer = f.read(RATE)
	if not len(buffer):  
		break
	else:
		buffers.append(buffer)

with socket.create_connection((hostname, 443)) as sock:
	with context.wrap_socket(sock, server_hostname=hostname) as ssock:
		requestLog(ssock,"POST " + endpoint + " HTTP/1.1\r\n")
		requestLog(ssock,"Host: " + hostname + "\r\n")
		requestLog(ssock,"Authorization: Bearer " + token + "\r\n")
		requestLog(ssock,f'Content-Type: audio/raw;encoding=signed-integer;bits={BIT_DEPTH};rate={RATE};endian=little\r\n')
		requestLog(ssock,"User-Agent: Mozilla/5.0\r\n")
		requestLog(ssock,"Transfer-Encoding: chunked\r\n")
		requestLog(ssock,"\r\n")

		for buffer in buffers:
			hexStr =hex(len(buffer))[2:]
			requestLog(ssock,hexStr)
			requestLog(ssock,"\r\n")
			ssock.sendall(buffer)
			print("<buffer content>")
			requestLog(ssock,"\r\n")

		requestLog(ssock,"0\r\n")
		requestLog(ssock,"\r\n")
```
So I wrote the "C++ version" of this code like this : 
```cpp
void startChunkedRequest(SSLClient client, String server, String endpoint)
{
  client.print(String("POST ") + endpoint + " HTTP/1.1\r\n");
  client.print(String("Host: ") + server + "\r\n");
  client.print("Authorization: Bearer " TOKEN "\r\n");
  client.print("Content-Type: audio/raw;encoding=signed-integer;bits=8;rate=50000;endian=little\r\n");
  client.print("User-Agent: Mozilla/5.0\r\n");
  client.print("Transfer-Encoding: chunked\r\n");
  client.print("\r\n");
}

void endChunkedRequest(SSLClient client)
{
  client.print("0\r\n");
  client.print("\r\n");
}
void chunkedRequest(SSLClient client, uint8_t *buffer, size_t len)
{
  client.print("c350");
  client.print("\r\n");
  client.write(buffer, len);
  client.print("\r\n");
}
```
But when I try it on my esp32 I get the error "connection reset by peer".
So if you know the differences between the two clients/code please let me know.
Thanks

prototypicalpro, 2020-09-18:
A connection reset generally means the server timed out the connection, which can be an indicator that SSLClient is not responding fast enough. One possible cause is that SSLClient's [write buffer](https://github.com/OPEnSLab-OSU/SSLClient#write-buffering) is never being flushed: do you call `client.flush()` or `client.available()` after calling the above functions?

MrMoine, 2020-09-19:
No I don't, I will read about those methodes and try to add them.
I'll keep you aware of the progress.
(thanks for the quick response btw ^^)

MrMoine, 2020-09-19:
Ok so I modified my `startChunkedRequest` like so : ```cpp
void startChunkedRequest(SSLClient client, String server, String endpoint)
{
  client.print(String("POST ") + endpoint + " HTTP/1.1\r\n");
  client.print(String("Host: ") + server + "\r\n");
  client.print("Authorization: Bearer " TOKEN "\r\n");
  client.print("Content-Type: audio/raw;encoding=signed-integer;bits=8;rate=50000;endian=little\r\n");
  client.print("User-Agent: Mozilla/5.0\r\n");
  client.print("Transfer-Encoding: chunked\r\n");
  client.print("\r\n");
  client.flush();
  Serial.println("Request started");
}```
But I don't know why `Request started` never prints, it seems like `SSLClient::flush` is blocking the function forever

prototypicalpro, 2020-09-21:
An SSLClient timeout usually means both the server and SSLClient are waiting on eachother to send data, which probably means you found a bug with `client.flush()`. Could you try placing the `client.flush()` call immediately after you finish your chunked request?

MrMoine, 2020-09-21:
I think that's what I did, [here ](https://pastebin.com/4MH0FGAD) is my new code for my functions. The `Serial.println()` at the end of the function `startChunkedRequest()` never prints.

prototypicalpro, 2020-09-21:
Here is what I mean:
```C++
void startChunkedRequest(SSLClient client, String server, String endpoint)
{
  client.print(String("POST ") + endpoint + " HTTP/1.1\r\n");
  client.print(String("Host: ") + server + "\r\n");
  client.print("Authorization: Bearer " TOKEN "\r\n");
  client.print("Content-Type: audio/raw;encoding=signed-integer;bits=8;rate=50000;endian=little\r\n");
  client.print("User-Agent: Mozilla/5.0\r\n");
  client.print("Transfer-Encoding: chunked\r\n");
  client.print("\r\n");
  // client.flush(); Remove for now
  Serial.println("Request started");
}

void endChunkedRequest(SSLClient client)
{
  client.print("0\r\n");
  client.print("\r\n");
  client.flush();
}
void chunkedRequest(SSLClient client, uint8_t *buffer, size_t len)
{
  client.print("c350");
  client.print("\r\n");
  client.write(buffer, len);
  client.print("\r\n");
  // client.flush(); Remove for now
}
```

My idea is to only call `client.flush()` when SSLClient should expect to receive data, which in this case should be when the request is 100% finished. If this change fixes the issue, it would mean that `client.flush()` has a bug where it prevents sending further data until data is received.

MrMoine, 2020-09-22:
I'm sorry, I forgot to mention that I get the error as soon as the `chunkedRequest()` function is called 😅 .

prototypicalpro, 2020-09-28:
That's interesting, in that case that solution will probably not work. Could you email me a copy of your sketch, and I can try and reproduce the problem? You can use the email on my GitHub profile.

prototypicalpro, 2020-09-30:
Actually I think I may have figured out the problem. In `chunkedRequest`:
```C++
void chunkedRequest(SSLClient client, uint8_t *buffer, size_t len)
```
should be
```C++
void chunkedRequest(SSLClient& client, uint8_t *buffer, size_t len)
```
The function interface at the top created a copy of SSLClient every function call, which is probably causing the unexpected behavior we are seeing. Instead, we would like to reference the existing instance of SSLClient using the function interface at the bottom.

MrMoine, 2020-10-01:
Sorry for my late answer I've been a little busy recently. I tried your solution but the problem remains, I have tried doing simple requests to troubleshoot and it seems that when I send the header `content-encoding:chunked`, `client.flush()` times out.
Do you think it can be related ?

MrMoine, 2020-10-03:
Ok so know I have managed to make my first function (`startChunkedRequest()`)  worked, with `client.flush()` that no longer times out. I fixed by replacing `client.println("something")` by `client.print("something\n\r")`.

But now I have these messages and errors when my function `sendChunk()` is called (the function used to send my audio buffers) : 
```(SSLClient)(SSL_WARN)(m_run_until): Discarded unread data to favor a write operation
(SSLClient)(SSL_WARN)(m_run_until): Discarded unread data to favor a write operation
(SSLClient)(SSL_WARN)(m_run_until): Discarded unread data to favor a write operation
(SSLClient)(SSL_WARN)(m_run_until): Discarded unread data to favor a write operation
(SSLClient)(SSL_WARN)(m_run_until): Terminating because the ssl engine closed
(SSLClient)(SSL_ERROR)(write): Failed while waiting for the engine to enter BR_SSL_SENDAPP
(SSLClient)(SSL_ERROR)(write): Cannot operate on a closed SSL connection.
(SSLClient)(SSL_ERROR)(write): Cannot operate on a closed SSL connection.
(SSLClient)(SSL_ERROR)(write): Cannot operate on a closed SSL connection.
After buffer
[D][WiFiClient.cpp:509] connected(): Disconnected: RES: -1, ERR: 104
```

prototypicalpro, 2020-10-04:
The `Discarded unread data to favor a write operation` warning occurs when SSLClient has buffered received data and the sketch calls `client.write()` before reading it. I suspect in this case what is happening is the server is returning an error before the SSLClient can finish sending the chunked request, causing SSLClient to buffer the response and then overwrite this buffer with more chunks.

Sorry I haven't been more responsive, I've been busier than usual lately. I think I have an idea of why this is failing, but I'll need to do some testing--I'll try to get back to you with results by the end of this week.

MrMoine, 2020-10-05:
That's actually so nice of you thank you.

prototypicalpro, 2020-10-05:
I'm still not sure what the problem you're experiencing is, but I was successfully been able to upload to `api.wit.ai/speech` using an ESP32 and SSLClient over WiFi. My sketch is shown below (the token has been revoked, but is left in for demo purposes):

<details>
<summary>WitAITest.ino</summary>

```C++
#include <SPI.h>
#include <WiFi.h>
#include <WiFiClient.h>
#include <SSLClient.h>
#include "trust_anchors.h"
#include "file.h"

// Enter a MAC address for your controller below.
// Newer Ethernet shields have a MAC address printed on a sticker on the shield
byte mac[] = { 0xDE, 0xAD, 0xBE, 0xEF, 0xFE, 0xED };

// Set the static IP address to use if the DHCP fails to assign
IPAddress ip(192, 168, 0, 177);
IPAddress myDns(8, 8, 8, 8);

// Choose the analog pin to get semi-random data from for SSL
// Pick a pin that's not connected or attached to a randomish voltage source
const int rand_pin = A5;

// Initialize the SSL client library
// We input an EthernetClient, our trust anchors, and the analog pin
WiFiClient base_client;
SSLClient client(base_client, TAs, (size_t)TAs_NUM, rand_pin, 1, SSLClient::SSL_INFO);
// Variables to measure the speed
unsigned long beginMicros, endMicros;
unsigned long byteCount = 0;
bool printWebData = true;  // set to false for better speed measurement

void connectToWiFi(const char * ssid, const char * pwd)
{
  Serial.println("Connecting to WiFi network: " + String(ssid));

  WiFi.begin(ssid, pwd);

  while (WiFi.status() != WL_CONNECTED) 
  {
    delay(500);
    Serial.print(".");
  }

  Serial.println();
  Serial.println("WiFi connected!");
  Serial.print("IP address: ");
  Serial.println(WiFi.localIP());
}

void startChunkedRequest(SSLClient& client)
{
  client.println("POST /speech HTTP/1.1");
  client.println("Host: api.wit.ai");
  client.println("Authorization: Bearer A35ZFUZCP2WZ4OIZDOBHWP3U3XNVLP6R");
  client.println("Content-Type: audio/raw;encoding=signed-integer;bits=16;rate=16000;endian=little");
  client.println("User-Agent: Mozilla/5.0");
  client.println("Transfer-Encoding: chunked");
  client.println();
}

void endChunkedRequest(SSLClient& client)
{
  client.println("0");
  client.println();
}
void chunkedRequest(SSLClient& client, const uint8_t *buffer, size_t len)
{
[... 2038 lines cut ...]
	0x01, 0x80, 0x04, 0x80, 0xFE, 0x7F, 0xFE, 0x7F, 0xFB, 0x7F, 0xFC, 0x7F, 0xFB, 0x7F, 0xFB, 0x7F, 
	0xFE, 0x7F, 0xFF, 0x7F, 0xFF, 0x7F, 0x02, 0x80, 0x04, 0x80, 0x05, 0x80, 0x03, 0x80, 0x08, 0x80, 
	0x07, 0x80, 0x02, 0x80, 0x07, 0x80, 0x03, 0x80, 0x02, 0x80, 0x04, 0x80, 0x01, 0x80, 0xFF, 0x7F, 
	0xFE, 0x7F, 0x01, 0x80, 0xFC, 0x7F, 0xFE, 0x7F, 0x03, 0x80, 0x05, 0x80, 0x06, 0x80, 0x07, 0x80, 
	0x09, 0x80, 0x09, 0x80, 0x08, 0x80, 0x06, 0x80, 0x04, 0x80, 0x02, 0x80, 0x01, 0x80, 0xFE, 0x7F, 
	0xFD, 0x7F, 0xFD, 0x7F, 0xF9, 0x7F, 0xF9, 0x7F, 0xF9, 0x7F, 0xF9, 0x7F, 0xF9, 0x7F, 0xFD, 0x7F, 
	0xFE, 0x7F, 0x03, 0x80, 0x02, 0x80, 0x03, 0x80, 0x06, 0x80, 0x05, 0x80, 0x07, 0x80, 0x04, 0x80, 
	0x03, 0x80, 0x03, 0x80, 0x01, 0x80, 0xFE, 0x7F, 0xFF, 0x7F, 0xFE, 0x7F, 0xFC, 0x7F, 0xFB, 0x7F, 
	0xFB, 0x7F, 0xFB, 0x7F, 0xFB, 0x7F, 0xFC, 0x7F, 0xFD, 0x7F, 0xFD, 0x7F, 0xFF, 0x7F, 0x00, 0x80, 
	0x02, 0x80, 0x01, 0x80, 0x02, 0x80, 0x04, 0x80, 0x00, 0x80, 0x02, 0x80, 0x01, 0x80, 0xFE, 0x7F, 
	0xFF, 0x7F, 0x01, 0x80, 0x02, 0x80, 0x03, 0x80, 0x03, 0x80, 0x02, 0x80, 0x03, 0x80, 0x03, 0x80, 
	0xFF, 0x7F, 0x02, 0x80, 0x02, 0x80, 0x00, 0x80, 0x03, 0x80, 0x01, 0x80, 0x01, 0x80, 0x00, 0x80, 
	0x03, 0x80, 0xFF, 0x7F, 0xFE, 0x7F, 0xFD, 0x7F, 0xFC, 0x7F, 0xFA, 0x7F, 0xF9, 0x7F, 0xFB, 0x7F, 
	0xF9, 0x7F, 0xFE, 0x7F, 0xFD, 0x7F, 0x00, 0x80, 0x04, 0x80, 0x03, 0x80, 0x03, 0x80, 0x04, 0x80, 
	0x03, 0x80, 0x03, 0x80, 0x02, 0x80, 0x00, 0x80, 0x01, 0x80, 0x00, 0x80, 0x00, 0x80, 0xFD, 0x7F, 
	0xFC, 0x7F, 0xFC, 0x7F, 0xFB, 0x7F, 0xFC, 0x7F, 0xFD, 0x7F, 0xFE, 0x7F, 0xFE, 0x7F, 0xFE, 0x7F, 
	0xFF, 0x7F, 0xFF, 0x7F, 0xFE, 0x7F, 0x00, 0x80, 0x03, 0x80, 0x01, 0x80, 0x00, 0x80, 0x02, 0x80, 
	0x05, 0x80, 0xFF, 0x7F, 0xFF, 0x7F, 0x01, 0x80, 0x00, 0x80, 0xFD, 0x7F, 0xFD, 0x7F, 0xFE, 0x7F, 
	0xFF, 0x7F, 0x00, 0x80, 0xFD, 0x7F, 0x04, 0x80, 0x01, 0x80, 0x02, 0x80, 0x05, 0x80, 0x02, 0x80, 
	0x02, 0x80, 0x01, 0x80, 0xFE, 0x7F, 0x00, 0x80, 0x00, 0x80, 0xFD, 0x7F, 0x02, 0x80, 0x00, 0x80, 
	0x02, 0x80, 0x02, 0x80, 0x07, 0x80, 0x00, 0x80, 0x06, 0x80, 0x05, 0x80, 0x00, 0x80, 0x05, 0x80, 
	0x00, 0x80, 0x04, 0x80, 0x05, 0x80, 0x02, 0x80, 0x03, 0x80, 0x08, 0x80, 0x07, 0x80, 0x05, 0x80, 
	0x08, 0x80, 0x05, 0x80, 0x02, 0x80, 0x00, 0x80, 0xFD, 0x7F, 0xFE, 0x7F, 0xFC, 0x7F, 0xFD, 0x7F, 
	0xFD, 0x7F, 0xFD, 0x7F, 0x00, 0x80, 0xFF, 0x7F, 0x00, 0x80, 0x02, 0x80, 0x00, 0x80, 0xFE, 0x7F, 
	0x00, 0x80, 0xFE, 0x7F, 0xFD,

MrMoine, 2020-10-07:
Ok I tried something and e-mailed it to you.

closed 2020-10-13 by prototypicalpro