OPEnSLab-OSU/SSLClient #17: Certificate to use AWS via MQTT

Mark-81, 2020-09-28:
I'm using on ESP8266 a sketch like this one: https://github.com/Ameba8195/Arduino/blob/master/hardware_v2/libraries/MQTTClient/examples/amazon_awsiot_basic/amazon_awsiot_basic.ino, where the certificates are simply stored in flash as they are downloaded from the server:

```
char* rootCABuff = \
"-----BEGIN CERTIFICATE-----\n" \
"MIIE0zCCA7ugAwIBAgIQGNrRniZ96LtKIVjNzGs7SjANBgkqhkiG9w0BAQUFADCB\n" \
"yjELMAkGA1UEBhMCVVMxFzAVBgNVBAoTDlZlcmlTaWduLCBJbmMuMR8wHQYDVQQL\n" \
...

char* certificateBuff = \
"-----BEGIN CERTIFICATE-----\n" \
"MIIDWTCCAkGgAwIBAgIUE1UsPqN2mfvCGh2DLX2HWs3NOIYwDQYJKoZIhvcNAQEL\n" \
"BQAwTTFLMEkGA1UECwxCQW1hem9uIFdlYiBTZXJ2aWNlcyBPPUFtYXpvbi5jb20g\n" \
...

char* privateKeyBuff = \
"-----BEGIN RSA PRIVATE KEY-----\n" \
"MIIEpAIBAAKCAQEA0zz9/MUl5mhLbIh/RjKx4WpSWfA3A2yDQbhT7eZQ+PjuiCze\n" \
"MsMUDbTw/zlLeqd8NpRjnnfBhjcFwiUHOmnLu+y2uBqlM7EfZz82uT9B8OqD+BQF\n" \
...
```

Because the STM32Ethernet library still does not support secure connections, they suggested me to use this library instead.
I read the documentation here: https://github.com/OPEnSLab-OSU/SSLClient/blob/master/TrustAnchors.md but my still limited knowledge about certificates doesn't allow me to fully understand how should I fill in the Trust Anchors from the above certificates.

prototypicalpro, 2020-09-28:
Hello! A trust anchor is BearSSL's way of storing the root CA (`rootCABuf`) with the extra information removed. The [documentation](https://bearssl.org/x509.html) talks about them in more detail, but the gist is that trust anchors must be generated from the appropriate root CAs using an external tool and then copied into the project. In your case, you can use [this python script included in SSLClient](https://github.com/OPEnSLab-OSU/SSLClient/blob/master/tools/pycert_bearssl/pycert_bearssl.py) to convert the certificate in `rootCABuf` into a trust anchor. Simply find this script in your copy of SSLClient (under `tools/pycert_bearssl/pycert_bearssl.py`), follow the setup instructions at the top of the file, and run `pycert_bearssl.py convert --no-search <certificate PEM file>`. This command should create a `certificates.h` file containing the trust anchors ([like so](https://github.com/OPEnSLab-OSU/SSLClient/blob/master/examples/EthernetMQTT/certificates.h)) which you can include in your project.

Sorry for the confusion, I need to revisit that documentation at some point.

Mark-81, 2020-09-29:
No problem, thanks for the clarification. Instead, the other certificates (`certificateBuff` and `privateKeyBuff` in my example) can be the same as above, and I would use `SSLClientParameters fromPEM();` to set them, as far as I understand.

prototypicalpro, 2020-09-29:
Excellent! I'll go ahead and close this issue so we can continue the discussion in https://github.com/OPEnSLab-OSU/SSLClient/issues/18.

closed 2020-09-29 by prototypicalpro