OPEnSLab-OSU/SSLClient #18: Disconnection while receiving a large message

Mark-81, 2020-09-29:
I'm not sure if this issue is related to `SSLClient` or to `PubSubClient`.
After subscribing to an MQTT topic, I can successfully receive messages. But if the message is "too large" I got a disconnection:

For example, the message I'm expecting from AWS (`$aws/things/<thing>/shadow/get`) is about 9 kbytes large. I've plenty of RAM on my STM32 so I added:

```
#define MQTT_PACKET_SIZE  16384

SSLClientParameters mTLS = SSLClientParameters::fromPEM(certificateBuff, sizeof certificateBuff, privateKeyBuff, sizeof privateKeyBuff);
EthernetClient ethClient;
SSLClient networkClient(ethClient, TAs, (size_t)TAs_NUM, A5);
PubSubClient mqtt(mqttServer, 8883, MQTT_Callback, networkClient);

void setup()
{
    ...
    mqtt.setBufferSize(MQTT_PACKET_SIZE);
    ...
}
```

My `MQTT_Callback` doesn't require a new buffer:

```
void MQTT_Callback(char *topic, byte *payload, unsigned int length)
{
  Serial.print("Message arrived [");
  Serial.print(topic);
  Serial.print("] ");
  for (int i=0;i<length;i++) Serial.print((char)payload[i]);
  Serial.println();
}
```

This is the output:

> 17:13:57.087 ->   DHCP assigned IP 192.168.1.179
> 17:13:57.087 -> [NET] MAC Address: 00:80:E1:3E:00:40
> 17:13:58.576 -> Attempting MQTT connection...connected
> 17:14:02.117 -> (SSLClient)(SSL_WARN)(m_run_until): Discarded unread data to favor a write operation
> 17:14:02.315 -> (SSLClient)(SSL_WARN)(m_run_until): Discarded unread data to favor a write operation
> 17:14:02.514 -> (SSLClient)(SSL_WARN)(m_run_until): Discarded unread data to favor a write operation
> 17:14:02.679 -> (SSLClient)(SSL_WARN)(m_run_until): Discarded unread data to favor a write operation
> 17:14:02.878 -> (SSLClient)(SSL_WARN)(m_run_until): Discarded unread data to favor a write operation
> 17:14:02.878 -> Published [$aws/things/<thing>/shadow/get] 
> 17:14:03.043 -> Attempting MQTT connection...(SSLClient)(SSL_WARN)(connect): Arduino client is already connected? Continuing anyway...
> 17:14:06.352 -> connected
> 17:14:06.518 -> (SSLClient)(SSL_WARN)(m_run_until): Discarded unread data to favor a write operation
> 17:14:06.716 -> (SSLClient)(SSL_WARN)(m_run_until): Discarded unread data to favor a write operation
> 17:14:06.882 -> (SSLClient)(SSL_WARN)(m_run_until): Discarded unread data to favor a write operation
> 17:14:07.080 -> (SSLClient)(SSL_WARN)(m_run_until): Discarded unread data to favor a write operation
> 17:14:07.246 -> (SSLClient)(SSL_WARN)(m_run_until): Discarded unread data to favor a write operation
> 17:14:07.279 -> Published [$aws/things/<thing>/shadow/get] 
> 17:14:07.411 -> Attempting MQTT connection...(SSLClient)(SSL_WARN)(connect): Arduino client is already connected? Continuing anyway...
> 17:14:10.753 -> connected
> 17:14:10.919 -> (SSLClient)(SSL_WARN)(m_run_until): Discarded unread data to favor a write operation
> 17:14:11.084 -> (SSLClient)(SSL_WARN)(m_run_until): Discarded unread data to favor a write operation
> ...
> 

I also tried to enable logging, and this is the new output:

>   DHCP assigned IP 192.168.1.179
> 17:18:45.719 -> [NET] MAC Address: 00:80:E1:3E:00:40
> 17:18:47.208 -> Attempting MQTT connection...(SSLClient)(SSL_INFO)(connect): Base client connected!
> 17:18:47.406 -> (SSLClient)(SSL_INFO)(m_run_until): m_run changed state:
> 17:18:47.406 -> (SSLClient)(SSL_INFO)(m_run_until): State: 
> 17:18:47.406 ->    RECVREC
> 17:18:47.406 -> (SSLClient)(SSL_INFO)(m_run_until): Expected bytes count: 
> 17:18:47.439 -> (SSLClient)(SSL_INFO)(m_run_until): 5
[... 221 lines cut ...]
> 17:18:57.234 -> (SSLClient)(SSL_INFO)(m_run_until): 5
> 17:18:57.234 -> (SSLClient)(SSL_INFO)(m_run_until): m_run changed state:
> 17:18:57.234 -> (SSLClient)(SSL_INFO)(m_run_until): State: 
> 17:18:57.234 ->    SENDAPP
> 17:18:57.234 -> (SSLClient)(SSL_INFO)(m_run_until): m_run changed state:
> 17:18:57.234 -> (SSLClient)(SSL_INFO)(m_run_until): State: 
> 17:18:57.267 ->    RECVREC
> 17:18:57.267 ->    SENDAPP
> 17:18:57.267 -> (SSLClient)(SSL_INFO)(m_run_until): Expected bytes count: 
> 17:18:57.267 -> (SSLClient)(SSL_INFO)(m_run_until): 5
> 17:18:57.400 -> (SSLClient)(SSL_INFO)(m_run_until): m_run changed state:
> 17:18:57.400 -> (SSLClient)(SSL_INFO)(m_run_until): State: 
> 17:18:57.400 ->    RECVAPP
> 17:18:57.400 -> (SSLClient)(SSL_INFO)(m_run_until): m_run changed state:
> 17:18:57.433 -> (SSLClient)(SSL_INFO)(m_run_until): State: 
> 17:18:57.433 ->    RECVAPP
> 17:18:57.433 -> (SSLClient)(SSL_WARN)(m_run_until): Discarded unread data to favor a write operation
> 17:18:57.433 -> (SSLClient)(SSL_INFO)(m_run_until): m_run changed state:
> 17:18:57.433 -> (SSLClient)(SSL_INFO)(m_run_until): State: 
> 17:18:57.433 ->    RECVREC
> 17:18:57.433 ->    SENDAPP
> 17:18:57.433 -> (SSLClient)(SSL_INFO)(m_run_until): Expected bytes count: 
> 17:18:58.458 -> (SSLClient)(SSL_INFO)(m_run_until): 5
> ...
>

Mark-81, 2020-09-29:
Hacking the library and setting:

    unsigned char m_iobuf[BR_SSL_BUFSIZE_BIDI];

leads to fix the disconnection as well as the warning "Discarded unread data to favor a write operation", but the published message is still not received.

prototypicalpro, 2020-09-29:
Could you post the logs with your modified version of the library?

Mark-81, 2020-09-30:
Here the full log:

> 09:19:35.474 -> [NET] DHCP assigned IP: 192.168.1.179
> 09:19:35.474 -> [NET] MAC Address: 00:80:E1:3E:00:40
> 09:19:35.474 -> [MQTT] Connecting...(SSLClient)(SSL_INFO)(connect): Base client connected!
> 09:19:35.706 -> (SSLClient)(SSL_INFO)(m_run_until): m_run changed state:
> 09:19:35.706 -> (SSLClient)(SSL_INFO)(m_run_until): State: 
> 09:19:35.739 ->    RECVREC
> 09:19:35.739 -> (SSLClient)(SSL_INFO)(m_run_until): Expected bytes count: 
> 09:19:35.739 -> (SSLClient)(SSL_INFO)(m_run_until): 5
> 09:19:38.717 -> (SSLClient)(SSL_INFO)(m_run_until): m_run changed state:
> 09:19:38.717 -> (SSLClient)(SSL_INFO)(m_run_until): State: 
> 09:19:38.717 ->    RECVREC
> 09:19:38.717 ->    SENDAPP
> 09:19:38.717 -> (SSLClient)(SSL_INFO)(m_start_ssl): Connection successful!
> 09:19:38.949 -> connected
> 09:19:38.982 -> (SSLClient)(SSL_INFO)(m_run_until): m_run changed state:
> 09:19:38.982 -> (SSLClient)(SSL_INFO)(m_run_until): State: 
> 09:19:38.982 ->    RECVREC
> 09:19:38.982 ->    SENDAPP
> 09:19:38.982 -> (SSLClient)(SSL_INFO)(m_run_until): Expected bytes count: 
> 09:19:38.982 -> (SSLClient)(SSL_INFO)(m_run_until): 5
> 09:19:39.148 -> (SSLClient)(SSL_INFO)(m_run_until): m_run changed state:
> 09:19:39.148 -> (SSLClient)(SSL_INFO)(m_run_until): State: 
> 09:19:39.148 ->    SENDAPP
> 09:19:39.148 ->    RECVAPP
> 09:19:39.148 -> [MQTT] Subscribed to: $aws/things/thing/shadow/update/#
> 09:19:39.181 -> (SSLClient)(SSL_INFO)(m_run_until): m_run changed state:
> 09:19:39.181 -> (SSLClient)(SSL_INFO)(m_run_until): State: 
> 09:19:39.181 ->    SENDAPP
> 09:19:39.181 ->    RECVAPP
> 09:19:39.181 -> [MQTT] Subscribed to: $aws/things/thing/shadow/get/#
> 09:19:39.181 -> [MQTT] Publish $aws/things/thing/shadow/get: 
> 09:19:39.181 -> [MQTT] Publish return code: 1

To be sure, on my development machine I subscribed to the same topics using `mosquitto_sub` (of course using a different client id) and after publishing the null message on `shadow/get` I receive the answer message there.

Please feel free to ask anything else I can do to help you to understand what's going on.

prototypicalpro, 2020-10-02:
It looks like this is an issue with PubSubClient not being able to handle a buffer size larger than 256 bytes: https://github.com/knolleary/pubsubclient/issues/764. If there were some way you could test your server without SSLClient we could confirm this. I'm going to assume that PubSubClient is the issue for now because the logs you posted look correct.

Unfortunately this doesn't leave us with a lot of good solutions to the problem. Glancing at the PubSubClient source, I noticed this code is assigning a `uint16_t` to a `uint8_t*` type: https://github.com/knolleary/pubsubclient/blob/2d228f2f862a95846c65a8518c79f48dfc8f188c/src/PubSubClient.cpp#L334. It's possible that this is causing an integer overflow and as a result the payload is being incorrectly read--if you are feeling particularly determined, you could attempt to fix that bug yourself and see if it improves the issue. Another possible option is changing the MQTT client you are using to a different one such as https://bitbucket.org/amotzek/mqtt-client/src/master/ or https://github.com/256dpi/arduino-mqtt.

Let me know what you end up doing, I'd love to incorporate your fix into the examples if you figure it out.

Mark-81, 2020-10-06:
Unfortunately I had not enough time to debug it... so I ended up to "trim" the published messages so they fit into a couple of kbytes and now all works fine. After the completion of the project I'm going to try again with larger messages.

prototypicalpro, 2021-03-25:
I'm going to go ahead and close this issue since it seems like an issue with PubSubClient (or at least an older version of it), feel free to re-open it if you find any indicators that the issue is with SSLClient.

closed 2021-03-25 by prototypicalpro