OPEnSLab-OSU/SSLClient #19: Email

gioreva, 2020-10-07:
Can you add one example for send email ?
Tls smtp
W5500
SAMD21

ps2chiper, 2022-01-01:
I succeeded in sending an email by SSLClient. I will write up a sketch and a tutorial on how to do it. I used an STM32F401CC and W5500 as my development environment. I don't even know if this is still needed?

gioreva, 2022-01-02:
Yes, need.
But for Android and without putting the certificates in the sources.
It must work with any domain.
Rather than not verify the certificate.

HebertWP, 2022-01-02:
it's not possible, Internet browsers don't requires  manual download of the certificates, it's only possible due the fact that it already in it source code of the navigator, along side it's "main certificates" they use a feature to find others certificates, using a trusted chain to download it, this operation it's extremely resources consuming. Andreas explain it munch better than me, watch his video https://youtu.be/Wm1xKj4bKsY

gioreva, 2022-01-02:
However it is possible not to check the certificate.

gioreva, 2022-01-02:
It was fine to send emails without TSL until it was possible.

ps2chiper, 2022-01-02:
I uploaded a super simple example of email working. I think it could work with less memory usage if I can figure out adequate memory constraints. 

[EmailSSL.zip](https://github.com/OPEnSLab-OSU/SSLClient/files/7799614/EmailSSL.zip)

ps2chiper, 2022-01-03:
I was able to get it to work on my Bluepill with 128KB flash.

I modified these two lines in the bearssl_ssl.h header.

/** \brief Optimal input buffer size. */
//#define BR_SSL_BUFSIZE_INPUT (16384 + 325)
#define BR_SSL_BUFSIZE_INPUT (1024 + 325)

/** \brief Optimal output buffer size. */
//#define BR_SSL_BUFSIZE_OUTPUT (16384 + 85)
#define BR_SSL_BUFSIZE_OUTPUT (1024 + 85)

It did fail once out of several email attempts from my Bluepill. I do not know the stability for using the connection on things other then simple text messages. I am just happy that after a year I was finally able to prove that it is possible to send an email from a Bluepill over SSL!

Sketch uses 103172 bytes (78%) of program storage space. Maximum is 131072 bytes.
Global variables use 13108 bytes (64%) of dynamic memory, leaving 7372 bytes for local variables. Maximum is 20480 bytes.

xreef, 2022-01-05:
Hi @gioreva,
Gmail supports an SSL certificate with an application password or a less secure app flag.
I will try to insert the code of ps2chiper inside my library to support GMail (like the other devices).
SSL also needs less memory.
Bye Renzo