OPEnSLab-OSU/SSLClient #20: mqtt.2030.ltsapis.goog:8883 generate/download does not work

fdistorted, 2020-10-13:
**Describe the bug**
I am trying connect SSL client to mqtt.2030.ltsapis.goog. I am having troubles with `pycert_bearssl.py`  

`python3 ./pycert_bearssl.py download --port 8883  mqtt.2030.ltsapis.goog` returns error  

I downloaded primary(https://pki.goog/gtsltsr/gtsltsr.crt) and backup(https://pki.goog/gsr4/GSR4.crt) certs from https://cloud.google.com/iot/docs/how-tos/mqtt-bridge#downloading_mqtt_server_certificates and converted them to PEM format.
`python3 ./pycert_bearssl.py convert --no-search ./primary.pem ./backup.pem` returns next error:  
```
Traceback (most recent call last):
  File "./pycert_bearssl.py", line 151, in <module>
    pycert_bearssl()
  File "/home/andrii/.platformio/penv/lib/python3.8/site-packages/click/core.py", line 829, in __call__
    return self.main(*args, **kwargs)
  File "/home/andrii/.platformio/penv/lib/python3.8/site-packages/click/core.py", line 782, in main
    rv = self.invoke(ctx)
  File "/home/andrii/.platformio/penv/lib/python3.8/site-packages/click/core.py", line 1259, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/home/andrii/.platformio/penv/lib/python3.8/site-packages/click/core.py", line 1066, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/home/andrii/.platformio/penv/lib/python3.8/site-packages/click/core.py", line 610, in invoke
    return callback(*args, **kwargs)
  File "./pycert_bearssl.py", line 147, in convert
    cert_util.x509_to_header(root_certs, cert_var, cert_length_var, output, keep_dupes)
  File "/home/andrii/Projects/DokoSens/firmware/scripts/cert_util.py", line 267, in x509_to_header
    n_bytes_str = bytes_to_c_data(numbers.n.to_bytes(pubkey.bits() // 8, byteorder="big"))
AttributeError: 'EllipticCurvePublicNumbers' object has no attribute 'n'
```  
As I understand provided above certs are  `EllipticCurve` instead of `RSA`. As `EllipticCurve` has `x` and `y` fields and not `n` and `e` script fails.

**Expected behavior**
That would be great to make this script generate correct `certificates.h` with different kinds of certificates. 

**Context (please complete the following information):**
 - Device Type ESP32 (LuaNode32)
 - Relevant Library Versions MQTT 2.4.7(https://libraries.io/platformio/MQTT)
 - SSLClient Version 1.6.8

prototypicalpro, 2020-10-13:
Unfortunately the `pycert_bearssl` tool doesn't support parsing elliptical curve certificates at the moment (mostly because I've never seen an ECC root in production until now). I'd be happy to add support for ECC, however I'm a bit busy at the moment and as a result it will take some time.

In the meantime, BearSSL provides a [command line tool](https://bearssl.org/gitweb/?p=BearSSL;a=tree;f=tools;h=0fa053e41d6bf88a28472f3b22dde41b21f14292;hb=HEAD) (`brssl`) for trust anchor generation which supports ECC certificates. To use this tool you must compile it from source (or find a release that someone has published), but once the tool is compiled the usage is [fairly straightforward](https://bearssl.org/api1.html#profiles).

fdistorted, 2020-10-15:
Thanks, I will try to use brssl. If I have some spare time I will try to do a PR to adjust `pycert_bearssl.py`

fdistorted, 2020-10-16:
I have build a BearSSL and tried to create a certificates.h file but brssl just converted certificates `base64` string into unsigned char* array. after some reverse engineering I managed achieve right format of certificates.h file to make it work in SSLClient. I created a pull request #21

prototypicalpro, 2020-10-16:
Looks great! I'll also go ahead and add these changes to https://openslab-osu.github.io/bearssl-certificate-utility/ so it can support EC certificates as well.

closed 2020-10-16 by prototypicalpro