OPEnSLab-OSU/SSLClient #29: Get certificate validity dates

samca208, 2020-12-11:
Is there a way to get the validity date of a certificate? Like the not after.....

prototypicalpro, 2020-12-23:
It depends on which certificate you would like the validity date from. For the trust anchors it isn't stored (the trust anchor is always assumed to be not expired) but it could be extracted when the trust anchors are generated and placed in a constant somewhere. For the certificate sent by the server to the device there isn't a readily available way to get that date in BearSSL, but you could feasibly extract it by inserting some code in the certificate validation engine. If it's just any certificate (such as a PEM string in memory) you could parse it using the [br_x509_decoder](https://bearssl.org/apidoc/bearssl__x509_8h.html#a1b34a7f2ce9f8c9086491551f848ad0f) API which will give you the validity date among other things.

Out of curiosity, what would you do with this date if you were able to get it?

samca208, 2020-12-24:
Figured it out and successful retrieved the not before, not after CN ecc ecc.
Reason: send valid to date via mqtt. If a newer certificate is present on server. Certificate path is forwarded to the esp32. Newer certificate is downloaded (server and client ID are mutually verified providing there own certificates)  thru a tls channel and written to firmware.

peter1a, 2021-01-18:
When using trust anchor is this permanent or should I update code every couple months when SSL expire?

prototypicalpro, 2021-01-19:
> When using trust anchor is this permanent or should I update code every couple months when SSL expire?
Yes, you should. Unfortunately the SSLClient doesn't have a way of updating it's own trust anchors, so they must instead be updated manually by regenerating and uploading again. The good news is that almost all certificates have a lifetime of a year or more, so you shouldn't need to update very frequently

peter1a, 2021-01-20:
> Yes, you should. Unfortunately the SSLClient doesn't have a way of updating it's own trust anchors, so they must instead be updated manually by regenerating and uploading again. The good news is [...]

What would be in this case proper way to update SW with new trust anchor if old stops working? Client can't connect to API which has SSL. In my case I have DST Root CA X3 which is valid until 30.9.2021.

peter1a, 2021-02-02:
Anyone have some idea?

peter1a, 2021-02-09:
Fix it with self-signed cert.

closed 2021-03-25 by prototypicalpro