OPEnSLab-OSU/SSLClient #32: How to convert Distinguished Name in hex

peter1a, 2021-02-03:
How is this hex code generated, can anybody explain me?

```
static const unsigned char TA_DN0[] = {
    0x30, 0x3f, 0x31, 0x24, 0x30, 0x22, 0x06, 0x03, 0x55, 0x04, 0x0a, 0x13,
    0x1b, 0x44, 0x69, 0x67, 0x69, 0x74, 0x61, 0x6c, 0x20, 0x53, 0x69, 0x67,
    0x6e, 0x61, 0x74, 0x75, 0x72, 0x65, 0x20, 0x54, 0x72, 0x75, 0x73, 0x74,
    0x20, 0x43, 0x6f, 0x2e, 0x31, 0x17, 0x30, 0x15, 0x06, 0x03, 0x55, 0x04,
    0x03, 0x13, 0x0e, 0x44, 0x53, 0x54, 0x20, 0x52, 0x6f, 0x6f, 0x74, 0x20,
    0x43, 0x41, 0x20, 0x58, 0x33,
};
```

If I convert this to ASCI my output is:
`0?1$0"U
Digital Signature Trust Co.10UDST Root CA X3`

Some strange characters inside string I don't know what it means.

Also I try to use https://openslab-osu.github.io/bearssl-certificate-utility/ but it won't work for my subdomain, keep saying that I need to add domain, so I need logic how in this case this application transform those informations into hex generated code.

prototypicalpro, 2021-02-05:
The DN is an ASN.1 sequence pulled directly from CA certificate. In other words, the DN can be extracted by parsing the CA certificate (encoded in [ASN.1](https://en.wikipedia.org/wiki/ASN.1) format and following the [X.509](https://en.wikipedia.org/wiki/X.509) specification), and extracting the byte sequence representing the certificate metadata according to the [BearSSL docs](https://bearssl.org/x509.html). ASN.1 is a binary encoding, which is why when you print out the raw DN bytes you get mostly garbage.

You can extract the DN manually, but it's a bit of a hassle and I don't recommend it. Instead, have you tried generating the trust anchor file using the included [pycert_bearssl](https://github.com/OPEnSLab-OSU/SSLClient/blob/master/tools/pycert_bearssl/pycert_bearssl.py) tool and the CA certificate for your domain (`pycert_bearssl convert cert.pem`)?

peter1a, 2021-02-05:
@prototypicalpro tnx for answring, found that pycert_bearssl tool works, but after running it on self-signed certificate it shows error: Could not find matching root certificate for domain: domain.com
Error: Could not download and/or validate the certificate for domain.com port 443!

Any thoughts about that?

prototypicalpro, 2021-02-05:
Right sorry the command should be `pycert_bearssl convert --no-search cert.pem`. The `--no-search` flag tells `pycert_bearssl` to use the provided certificate as the root certificate instead of attempting to find a matching root certificate, which should be what you need.

prototypicalpro, 2021-03-25:
I'm going to go ahead and close this issue, feel free to re-open it if that didn't solve your problem!

closed 2021-03-25 by prototypicalpro