OPEnSLab-OSU/SSLClient #34: Dynamic Certificates

genotix, 2021-03-24:
**AWS IoT would like to see a certificate -per device-**
It would be nice to be able to provision devices with their unique certificate on the fly.
Either by storing the certificate in EEPROM or on filesystem (I'm using **LittleFS** at this stage).

The SSLClientParameters::fromPEM() is used to provision the certificate information to the instance but at the moment of execution LittleFS is not approachable (this will become approachable once setup() starts; I'm using ArduinoIDE here)

**Possible solution**
Providing two reserved char arrays that will be filled once LittleFS has started in order to read out the certificates from the filesystem and put them in memory. I would like to provide a hard-coded "fallback" certificate and key which I could use to onboard devices.

Being able to call functions that change the certificate of the global instance would also be fine and probably even more flexible and generic.
I might be completely missing existing option due to lack of knowledge or general incompetence (one of my strong suits).
Having an example that shows how to deal with this would be great!

MaffooClock, 2021-03-24:
Interesting coincidence -- I was looking for _exactly_ this solution.

I've found that if I call `SSLClientParameters::fromPEM()` from within a function (which determines whether to use a saved certificate from memory or a default one compiled in), the MCU hangs -- seems like it only works when called in the global scope.

genotix, 2021-03-24:
> Interesting coincidence -- I was looking for _exactly_ this solution. I, too, want my device to dynamically retrieve a new certificate from AWS IoT service to store and use locally, which means the initial mTLS connection uses a static, compiled-in provisioning certificate.
> 
> I've found that if I call `SSLClientParameters::fromPEM()` from within a setup function (which determines whether to use a saved certificate from memory or a default one compiled in), the MCU hangs -- seems like it only works when called in the global scope.

I believe it will work but only for the length of the existence of the variable lifetime.
I haven't found it to be stable though (which is likely due to my lack of experience) and would think this as a well-embraced feature.

MaffooClock, 2021-03-24:
> I believe it will work but only for the lifetime of the variable.
> When your setup() function ends; so will the declaration of this variable.
I know all about variable scope, so let's skip the elementary stuff.  I'm saying the same thing you are, just adding supportive commentary.

This works fine:
```c++

SSLClient sslClient( gsmClient, TAs, TAs_NUM, A6 );
SSLClientParameters mTLS = SSLClientParameters::fromPEM( EMBEDDED_CERTIFICATE, sizeof EMBEDDED_CERTIFICATE, EMBEDDED_PRIVATEKEY, sizeof EMBEDDED_PRIVATEKEY );

void setup()
{
    sslClient.setMutualAuthParams( mTLS );
}

void main()
{
  // Make a mTLS connection (e.g. MQTT client) on sslClient object.
}
```

This does not:
```c++

SSLClient sslClient( gsmClient, TAs, TAs_NUM, A6 );

void setup()
{
    SSLClientParameters mTLS = SSLClientParameters::fromPEM( EMBEDDED_CERTIFICATE, sizeof EMBEDDED_CERTIFICATE, EMBEDDED_PRIVATEKEY, sizeof EMBEDDED_PRIVATEKEY );
    sslClient.setMutualAuthParams( mTLS );
}

void main()
{
  // Make a mTLS connection (e.g. MQTT client) on sslClient object.
  // This is where the MCU hangs
}
```

Because of this behavior, this makes it impossible to set the mTLS parameters dynamically (just as @genotix reported).

genotix, 2021-03-24:
Clear; Exactly! Thanks for the example.

prototypicalpro, 2021-03-24:
Hello @MaffooClock! Your second example doesn't work because SSLClient doesn't copy `mTLS`, which causes your code to use a dangling reference to the `mTLS` object on the stack and probably crash. This behavior is by design, as the `mTLS` object is very large and copying it would be too much for the lower-end microcontrollers SSLClient supports. This doesn't mean that changing the mutual authentication parameters dynamically is impossible—it does mean, however, that the lifetime of the mutual authentication parameters needs to be as long as the time they are active in SSLClient. One method of accomplishing this is using a global `unique_ptr`:
```C++
SSLClient sslClient( gsmClient, TAs, TAs_NUM, A6 );
std::unique_ptr<SSLClientParameters> mTLS;

void setup()
{
  // note that the paramterers to SSLClient::fromPEM are copied, so it is safe to destroy them afterwards
  mTLS = std::unique_ptr<SSLClientParameters>(new SSLClientParameters(SSLClientParameters::fromPEM(...)));
  sslClient.setMutualAuthParams(*mTLS);
  ...
}

void main()
{
  // you can also change certificates on the fly with this approach:
  if (...) {
    Serial.println("Changing certificates!");
    sslClient.stop();
    mTLS = std::unique_ptr<SSLClientParameters>(new SSLClientParameters(SSLClientParameters::fromPEM(...)));
    sslClient.setMutualAuthParams(*mTLS);
    reconnect();
    Serial.println("Changing certs completed!");
  }
}
```
I tested the above example with my ESP32, and I was successfully able to change certificates and reconnect.

genotix, 2021-03-24:
@prototypicalpro thanks for your example!
I will need to let that code sink in a bit...
It’s beyond what I have learnt.

MaffooClock, 2021-03-24:
NICE.  `std::unique_ptr<SSLClientParameters> mTLS;` is exactly the magic sauce I was trying to figure out, since having an uninitialized `mTLS` obviously wouldn't work.

closed 2021-03-25 by prototypicalpro

bvernham, 2022-08-29:
I tried the above with and MKR1010 and I got multiple error on compiling
'unique_ptr' in namespace 'std' does not name a template type

It highlighted the below line:

std::unique_ptr<SSLClientParameters> mTLS;

AWS_IoT_WiFi_SSLClient:48:6: error: 'unique_ptr' in namespace 'std' does not name a template type
 std::unique_ptr<SSLClientParameters> mTLS;
      ^~~~~~~~~~
C:\Users\bvernham\Dropbox (Isuzu North America)\Personal Backup\Arduino\MKR1010\AWS_IoT_WiFi_SSLClient\AWS_IoT_WiFi_SSLClient.ino: In function 'void setup()':
AWS_IoT_WiFi_SSLClient:56:3: error: 'mTLS' was not declared in this scope
   mTLS = std::unique_ptr<SSLClientParameters>(new SSLClientParameters(SSLClientParameters::fromPEM(SECRET_CERTIFICATE, sizeof SECRET_CERTIFICATE, SECRET_KEY, sizeof SECRET_KEY)));
   ^~~~
AWS_IoT_WiFi_SSLClient:56:15: error: 'unique_ptr' is not a member of 'std'
   mTLS = std::unique_ptr<SSLClientParameters>(new SSLClientParameters(SSLClientParameters::fromPEM(SECRET_CERTIFICATE, sizeof SECRET_CERTIFICATE, SECRET_KEY, sizeof SECRET_KEY)));
               ^~~~~~~~~~
AWS_IoT_WiFi_SSLClient:56:45: error: expected primary-expression before '>' token
   mTLS = std::unique_ptr<SSLClientParameters>(new SSLClientParameters(SSLClientParameters::fromPEM(SECRET_CERTIFICATE, sizeof SECRET_CERTIFICATE, SECRET_KEY, sizeof SECRET_KEY)));
                                          ^
'unique_ptr' in namespace 'std' does not name a template type

simogaspa84, 2022-12-23:
Hi @prototypicalpro .. I am trying to do what you suggested .. 

`mTLS = std::unique_ptr<SSLClientParameters>(new SSLClientParameters(SSLClientParameters::fromPEM(my_cert, sizeof my_cert, my_key, sizeof my_key)));`

but I got  

`error: no match for 'operator=' (operand types are 'SSLClientParameters' and 'std::unique_ptr<SSLClientParameters>')`

any idea about what could be ? 

Thanks