OPEnSLab-OSU/SSLClient #35: MQTT mTLS problem

nickehallgren, 2021-03-29:
Hi and thanks for this great software!

I'm trying to to use SSLClient to connect to my local mosquitto server with selfsigned certs but I can't figure out whats wrong. I already looked at [#7 ](https://github.com/OPEnSLab-OSU/SSLClient/issues/7) and I think I've done everything right (but then it would work, so I'm missing something).

I'm on a SAMD21 M0 Mini with a W5500 ethernet adapter, using the EthernetMQTT example with these changes (changed all Serial.print... to SerialUSB.print. Had to do that in SSLClient.cpp and SSLClient.h also):

```
const char my_cert[] = 
("-----BEGIN CERTIFICATE-----\r\n"
"MIIFITCCBAmgAwIBAgIJAO0g55P8M5rqMA0GCSqGSIb3DQEBDQUAMGAxGTAXBgNV\r\n"
"BAMMEEhvbWUgTVFUVCBicm9rZXIxFDASBgNVBAoMC2hhbGxncmVuLmZpMQswCQYD\r\n"
"VQQLDAJOQTEgMB4GCSqGSIb3DQEJARYRbmlja2VAaGFsbGdyZW4uZmkwHhcNMTgx\r\n"
"MDI3MTE0NjAyWhcNMzIxMDIzMTE0NjAyWjBeMRcwFQYDVQQDDA5ob21lYXV0b21h\r\n"
"dGlvbjEUMBIGA1UECgwLaGFsbGdyZW4uZmkxCzAJBgNVBAsMAk5BMSAwHgYJKoZI\r\n"
"hvcNAQkBFhFuaWNrZUBoYWxsZ3Jlbi5maTCCASIwDQYJKoZIhvcNAQEBBQADggEP\r\n"
"ADCCAQoCggEBAK9wUAhKjkDiu871VR+FhBv9Txl7iqdYT2WY/f7Vj6VBbozhde8V\r\n"
"N8uKB6Jm5xaVBo+LLtbXlUSAEdNt0kiJgH9ycXwHOMki17wEezix61VFmvnSVzOf\r\n"
"bFrWVIRfWKbMVZObPUy6Y2QN0Kiu4Sbk6KXVRAxZQRnFnwMWp3OfeocA1+Tpcup9\r\n"
"Jg/ZHIhSVPNRe6H3sEcRNZc2/DdvBwBGr50mIFqKBSu+3h+x3MXMySSEHwnxnSvH\r\n"
"W06aWpCqTXNiQxHYUh/tu/Dz+VVJrafyRhxv4VI+uaDUZqZTHh3Y5NFrMNbrBz6/\r\n"
"3Er1H28SAnNeWGLS0WaM/BIFybBRI0p0TwcCAwEAAaOCAd4wggHaMAwGA1UdEwEB\r\n"
"/wQCMAAwEQYJYIZIAYb4QgEBBAQDAgZAMAsGA1UdDwQEAwIF4DAhBglghkgBhvhC\r\n"
"AQ0EFBYSQnJva2VyIENlcnRpZmljYXRlMB0GA1UdDgQWBBSAz2y4oL4ARoAswHK+\r\n"
"O7NWPNUU/DCBkgYDVR0jBIGKMIGHgBQb9pcbqsJG+nyMDccdo9Eki47Y36FkpGIw\r\n"
"YDEZMBcGA1UEAwwQSG9tZSBNUVRUIGJyb2tlcjEUMBIGA1UECgwLaGFsbGdyZW4u\r\n"
"ZmkxCzAJBgNVBAsMAk5BMSAwHgYJKoZIhvcNAQkBFhFuaWNrZUBoYWxsZ3Jlbi5m\r\n"
"aYIJAIfLPF8FuBU8MEoGA1UdEQRDMEGHBMCoARKHEP6AAAAAAAAAs0vEHTLdZWCH\r\n"
"BH8AAAGHEAAAAAAAAAAAAAAAAAAAAAGHBMCoARKCCWxvY2FsaG9zdDCBhgYDVR0g\r\n"
"BH8wfTB7BgMrBQgwdDAcBggrBgEFBQcCARYQaHR0cDovL2xvY2FsaG9zdDBUBggr\r\n"
"BgEFBQcCAjBIMBAWCU93blRyYWNrczADAgEBGjRUaGlzIENBIGlzIGZvciBhIGxv\r\n"
"Y2FsIE1RVFQgYnJva2VyIGluc3RhbGxhdGlvbiBvbmx5MA0GCSqGSIb3DQEBDQUA\r\n"
"A4IBAQCPNpR9kNGmgHjSQmOg3iSrI5eykeie350LM9HIDLJ2Q2fzNDzUS+UQOBpI\r\n"
"1OR1ByyAndL58kGOM+9cyBVVpWOIVWsbCKSORLBDE/1OIvxCZyomDyEF3wuBjoGH\r\n"
"2jJGbOHjiX2Ij3gYTB63jFXyLDTeaEErUSz4jqfSrg/eId7q0OQbKOWUOaUrWty5\r\n"
"8QvrippU8qNKUJWf4RtlTqRyFCm2BUIoHol/S1jRopYiI67MVu3AcjFoRhm3HMyS\r\n"
"t9MOr6lKbkvw2CQdWLe5ml8Dh+tvsW1lPyb9ee4DHHtcMBro7egLsLBI/wcSnaGW\r\n"
"3iJGJ0K6dXZFICqjoyqwr718SXx3\r\n"
"-----END CERTIFICATE-----\r\n");
const char my_key[] = 
("-----BEGIN RSA PRIVATE KEY-----\n"
"MIIEpAIBAAKCAQEAr3BQCEqOQOK7zvVVH4WEG/1PGXuKp1hPZZj9/tWPpUFujOF1\r\n"
"7xU3y4oHombnFpUGj4su1teVRIAR023SSImAf3JxfAc4ySLXvAR7OLHrVUWa+dJX\r\n"
"M59sWtZUhF9YpsxVk5s9TLpjZA3QqK7hJuTopdVEDFlBGcWfAxanc596hwDX5Oly\r\n"
"6n0mD9kciFJU81F7ofewRxE1lzb8N28HAEavnSYgWooFK77eH7HcxczJJIQfCfGd\r\n"
"K8dbTppakKpNc2JDEdhSH+278PP5VUmtp/JGHG/hUj65oNRmplMeHdjk0Wsw1usH\r\n"
"Pr/cSvUfbxICc15YYtLRZoz8EgXJsFEjSnRPBwIDAQABAoIBAA0qSnc00RQBb8KK\r\n"
"GocxB8bp7WOmJ7ODJwIixvy2nf7yuA8OZEE6wGMfyo7fVfwV1tYyxaMNrn/jdHL/\r\n"
"TMZwuxnVRrrd997wJGxRkcWhZF2TyZtxs2WGUyVF7DW6yfZKYymefq0hV5AMkVn3\r\n"
"NaIrj2HSbTbYA9ChSdt7ebltgE5dG//huKI9NOMdE5bfWr4c/ZLUG063I48KnAZl\r\n"
"wlM7dioutO9Zl7u04aPNh76v/km9XihdjgyWAbPozPc7t30n0zEdq0IHTNRgEAUC\r\n"
"MSPavmZVqreqt5DI9NXHwytlj6pGqUUQVxHUAgrqzAdZo8gLGbR3UQL2T90E/Sna\r\n"
"Zq+UUnECgYEA3YN0Nqy/xnVISRx5ayF569H7+nglRiXeEyF9GPVtrDgyoEBhFzeS\r\n"
"SmSoeRlDBHCwHRcCCdFIAxWc6iiN3JjlarWTr+RceK7U39eFIywvJDOaxvgOcMjY\r\n"
"v/dyap0bAsf5OnS1SoGRnqLiPXHx1XHiWDERm0yV8Cpv/F6nG83El6UCgYEAysCF\r\n"
"wvVKEWcI0zc5VjkhY+N3zJCkZjhJp8ILtfsjez2NoA9oy4fe18f9O3op8FZbj4yC\r\n"
"WGvhZhLBosr6QYsW5+1z+605P1Gw7vWwO3aqwhMYt8wci0pEBocT32A44fD2g8aW\r\n"
"93JgHXBvzSjeN4mBvWFHVCyIBmv3IgTAYl+xLDsCgYEApKtHJinveonPTcJkJJbD\r\n"
"OuJj7kOIeghGeXRIudghSoiQYf7Z7ld6YeFIOJXl8xQOZ+Glu0SfwuM7Pfsd0mK7\r\n"
"EUOgYX5im7hkOW7O42HCPag7JDRdD8zPDI3XuZfG759XD/SYnKWLaiFuC/17qI+7\r\n"
"kpIPo6IgSGRT0YlPsuSZg50CgYEAsrbGcNVTR0mhrsOjp4vOEcUD6O2p62w8CEHz\r\n"
"nqHSYaQc/PP4NEb8NwlR/iYtZ2pHZJ2+mv/ketuXkM4HjtRyCVb1f+btahGwusEX\r\n"
"aSKtW0oH5KJmmqSIl5RjAOdI3GdQc1EN6ukHLYvZEGawkRWEpZBklh52e+qYuISl\r\n"
"q1Cb09cCgYAsDhuoOMWWXbczhW5HW+Xk6DQydcDRwBQtPaiTPScxZUS2pMdug6Pt\r\n"
"5glzQDndmc0/MyWw3tJF3EFEf32d0n6pCeyX09xdZ+ZzKU3Z58luJxOE0RJKmWt4\r\n"
"PUiDJkZVRtd/aIhLt08LAVHV9aKRt7Lu/ub6gB7c0bC0lirF2QIlXQ==\r\n"
"-----END RSA PRIVATE KEY-----\r\n");
SSLClientParameters mTLS = SSLClientParameters::fromPEM(my_cert, sizeof my_cert, my_key, sizeof my_key);

byte mac[] = {  0xDE, 0xED, 0xBA, 0xFE, 0xFE, 0xED };

IPAddress mqttServer(10, 10, 10, 10);
const int mqttPort = 8883;

EthernetClient ethClient;
SSLClient ethClientSSL(ethClient, TAs, (size_t)TAs_NUM, A0, 1, SSLClient::SSL_DUMP);
PubSubClient client(mqttServer, mqttPort, callback, ethClientSSL);

```
I generated my certificates.h like this:

```
openssl s_client -showcerts -connect 10.10.10.10:8883 </dev/null 2>/dev/null|openssl x509 -outform PEM >mycertfile.pem
python3 ./Arduino/libraries/SSLClient/tools/pycert_bearssl/pycert_bearssl.py convert --no-search mycertfile.pem
```

And this is the output from the serial monitor;

```
Attempting MQTT connection...(SSLClient)(SSL_WARN)(connect): Using a raw IP Address for an SSL connection bypasses some important verification steps. You should use a domain name (www.google.com) whenever possible.
(SSLClient)(SSL_INFO)(connect): Base client connected!
(SSLClient)(SSL_INFO)(m_print_br_state): (SSLClient)(SSL_INFO)(m_print_br_state): State: 
   SENDREC
(SSLClient)(SSL_INFO)(m_print_br_state): (SSLClient)(SSL_INFO)(m_print_br_state): State: 
   RECVREC
(SSLClient)(SSL_INFO)(m_run_until): m_run changed state:
(SSLClient)(SSL_INFO)(m_print_br_state): (SSLClient)(SSL_INFO)(m_print_br_state): State: 
   RECVREC
(SSLClient)(SSL_INFO)(m_run_until): Expected bytes count: 
(SSLClient)(SSL_INFO)(m_run_until): 5
(SSLClient)(SSL_INFO)(m_print_br_state): (SSLClient)(SSL_INFO)(m_print_br_state): State: 
   Connection closed
(SSLClient)(SSL_WARN)(m_run_until): Terminating because the ssl engine closed
(SSLClient)(SSL_ERROR)(m_start_ssl): Failed to initlalize the SSL layer
(SSLClient)(SSL_ERROR)(m_print_br_error): Chain could not be linked to a trust anchor. See https://github.com/OPEnSLab-OSU/SSLClient/blob/master/TrustAnchors.md
failed, rc=-2 try again in 5 seconds
```

I've also tried with only \n  my_cer and my_key. Any ideas on what I'm doing wrong?

prototypicalpro, 2021-03-30:
Hello @nickehallgren! Your client certificate and private key (`my_cert` and `my_key`) look good, though you'll want to regenerate them since you posted them online. I'm pretty sure the problem is with either the way you are generating the trust anchors or the certificate you are generating trust anchors with. Could you post `mycertfile.pem` so I can test with it? You can also email me it directly if you're not comfortable posting it online.

andrewmartinjames, 2021-03-30:
Hi @prototypicalpro @nickehallgren - I am having a nearly identical problem, except that I am trying to use Google IoT Core, with the CA certificate of mqtt.2030.ltsapis.goog:8883 which I generated exactly the same way as Nick. I'm attaching my certificates.h file as well as the pem (since it just has publicly available information) in the hopes it can help narrow down the issue for everyone. 
[certificates.h.zip](https://github.com/OPEnSLab-OSU/SSLClient/files/6232326/certificates.h.zip)
[mycertfile.pem.zip](https://github.com/OPEnSLab-OSU/SSLClient/files/6232331/mycertfile.pem.zip)

prototypicalpro, 2021-03-31:
I've figured out the issue: the certificates you're generating trust anchors with are _end entity_ certificates and not _certificate authorities_. In other words, you're generating the trust anchors using the bottom of the chain instead of the top, which is why SSLClient fails to associate the certificate being sent by the server with the provided trust anchor. There are two possible solutions to this problem:
 * The first is to change which certificate you're generating the trust anchors from to the relevant CA certificate. For Google IoT core, you can find which certificates they use [here](https://cloud.google.com/iot/docs/how-tos/mqtt-bridge#downloading_mqtt_server_certificates) (under the minimal CA set) and I've attached the generated trust anchor files for them [here](https://github.com/OPEnSLab-OSU/SSLClient/files/6239139/certificates.zip). @nickehallgren if you generated your own certificate chain you should have the CA on hand, if not you may need to do some poking around.
 * If you are not able to acquire the CA for whatever reason, another solution is telling BearSSL that the certificate is _not_ a CA and should be treated as such. You can do this by removing the `BR_X509_TA_CA` flag in the trust anchor structure, shown below. I can also push an update to `pycert_bearssl` so it does this automatically. Note that this solution probably won't work for Google IoT core as they most certainly have more than one possible end-entity certificate that the device will need to validate, and with this solution SSLClient can only validate the certificate provided.
```C++
...
static const br_x509_trust_anchor TAs[] = {
    {
        { (unsigned char *)TA_DN0, sizeof TA_DN0 },
        BR_X509_TA_CA, // <-- replace this with a zero
        {
            BR_KEYTYPE_RSA,
            { .rsa = {
                (unsigned char *)TA_RSA_N0, sizeof TA_RSA_N0,
                (unsigned char *)TA_RSA_E0, sizeof TA_RSA_E0,
            } }
        }
    },
};
...
```

Does that help?

nickehallgren, 2021-03-31:
@prototypicalpro I have the ca.crt, I tried like this and it works...

`python3 ./Arduino/libraries/SSLClient/tools/pycert_bearssl/pycert_bearssl.py convert --no-search ca.crt`

Thank you so much!!!

closed 2021-03-31 by nickehallgren

roysG, 2022-01-11:
@nickehallgren  Thanks!