OPEnSLab-OSU/SSLClient #36: vector: No such file or directory

javierpal, 2021-03-30:
**Describe the bug**
when trying to compiles it does not compile, I don't know if i need another lib or something, I only installed this library and try to compile but is not working

**To Reproduce**
Write some steps to reproduce the behavior. Be sure to include the vant hardware required, if any.
1. Compile using the library

**Expected behavior**
Compile fine

**Screenshots/Serial Output**
In file included from F:\Bibliotecas\Documentos\Arduino\libraries\SSLClient\src/SSLClient.h:23:0,
                 from F:\Bibliotecas\Documentos\Arduino\jesusMQTT_4relaysNANO\jesusMQTT_4relaysNANO.ino:4:
F:\Bibliotecas\Documentos\Arduino\libraries\SSLClient\src/SSLClientParameters.h:31:10: fatal error: vector: No such file or directory
 #include <vector>
          ^~~~~~~~
compilation terminated.
exit status 1
Error compilando para la tarjeta Arduino Nano.

I'm Using Arduino Nano with Ethernet Shield.

cotestatnt, 2021-03-30:
Arduino Nano is not a board supported.
I believe ATmega328 MCU doesn't have enough computing power to handle an SSL / TLS connection.

prototypicalpro, 2021-03-30:
> Arduino Nano is not a board supported.

Yup, unfortunately SSLClient won't fit on any AVR boards (Arduino Uno, Nano, Mega, etc.) as the library itself is too large. You'll need to either use unencrypted communication or switch to a more powerful microcontroller.

closed 2021-03-30 by prototypicalpro