OPEnSLab-OSU/SSLClient #38: AWS MQTT, wifi ESP32

HebertWP, 2021-04-01:
**i was look ways to use this library with SIM868, and integrate with tinyGsm Library, but before star the test with the GPRS properly, i have decided test with the native wifi in ESP32,easy to test :)**

The problem is,  when i use the native library(WiFiClientSecure) it's work well, but using SSLClient its say that the is write erro 

To test i have  this code 
```
#define SSL

#include <WiFi.h>
#include <WiFiClientSecure.h>
#include <PubSubClient.h>
#include "PassWordFile.h"
#include <SSLClient.h>
#include "certificates.h"
#include "cert.h"
#include "CA.h"

void pubSubCheckConnect(PubSubClient pubSubClient);

#if defined(SSL)
SSLClientParameters mTLS = SSLClientParameters::fromPEM(certificate_pem_crt, sizeof certificate_pem_crt, private_pem_key, sizeof private_pem_key);
WiFiClient clientnoSSL;
SSLClient client(clientnoSSL, TAs, (size_t)TAs_NUM, 27);
#else
WiFiClientSecure client;
#endif

PubSubClient pubSubClient(ENDPOINTAWS, 8883, client);

void setup()
{
     Serial.begin(115200);
    delay(50);
    Serial.printf("\r\nESP32 AWS IoT Example\r\n");
    Serial.printf("SDK version: %s\r\n", ESP.getSdkVersion());
    Serial.print("Connecting to ");
    Serial.print(WIFI_NAME);
    WiFi.begin(WIFI_NAME, WIFI_PASS);
    WiFi.waitForConnectResult();
    Serial.print(", WiFi connected, IP address: ");
    Serial.println(WiFi.localIP());
    Serial.printf("free heap :%i\r\n",esp_get_free_heap_size());
    pubSubClient.setBufferSize(2048);
#if !defined(SSL)
    client.setCACert(rootCA);
    client.setCertificate(certificate_pem_crt);
    client.setPrivateKey(private_pem_key);
#endif
}

unsigned long lastPublish;
int msgCount;

void loop()
{
    while (!pubSubClient.connected())
        pubSubClient.connect("ESPthingXXXX");
    pubSubClient.loop();
    client.flush();
    if (millis() - lastPublish > 10000)
    {
        String msg = String("{\"Hello from ESP32\": \"") + ++msgCount + String("\"}");
        boolean rc = pubSubClient.publish("outTopic", msg.c_str());
        client.flush();
        Serial.print("Published, rc=");
        Serial.print((rc ? "OK: " : "FAILED: "));
        Serial.println(msg);
        lastPublish = millis();
    }
}

void pubSubCheckConnect(PubSubClient pubSubClient)
{
    if (!pubSubClient.connected())
    {
        Serial.print("PubSubClient connecting to: ");
        Serial.print(ENDPOINTAWS);
        while (!pubSubClient.connected())
        {
            Serial.print(".");
            pubSubClient.connect("ESPthingXXXX");
            delay(1000);
        }
        Serial.println(" connected");
        pubSubClient.subscribe("inTopic");
    }
    pubSubClient.loop();
}
```
when the first line is on, my output is:
```

ESP32 AWS IoT Example
SDK version: v3.2.3-14-gd3e562907
Connecting to 350, WiFi connected, IP address: 192.168.1.105
free heap :201884
(SSLClient)(SSL_ERROR)(m_run_until): SSL internals timed out! This could be an internal error, bad data sent from the server, or data being discarded due to a buffer overflow. If you are using Ethernet, did you modify the library properly (see README)?
(SSLClient)(SSL_ERROR)(connected): Not connected because write error is set
(SSLClient)(SSL_ERROR)(m_print_ssl_error): SSL_BR_WRITE_ERROR
(SSLClient)(SSL_ERROR)(m_start_ssl): Failed to initlalize the SSL layer
(SSLClient)(SSL_ERROR)(m_print_br_error): Unknown error code: 0
(SSLClient)(SSL_ERROR)(connected): Not connected because write error is set
(SSLClient)(SSL_ERROR)(m_print_ssl_error): SSL_BR_WRITE_ERROR
(SSLClient)(SSL_ERROR)(connected): Not connected because write error is set
(SSLClient)(SSL_ERROR)(m_print_ssl_error): SSL_BR_WRITE_ERROR
(SSLClient)(SSL_ERROR)(connected): Not connected because write error is set
(SSLClient)(SSL_ERROR)(m_print_ssl_error): SSL_BR_WRITE_ERROR
```
with first line commented:
``` 
ESP32 AWS IoT Example
SDK version: v3.2.3-14-gd3e562907
Connecting to 350, WiFi connected, IP address: 192.168.1.105
free heap :211080
Published, rc=OK: {"Hello from ESP32": "1"}
Published, rc=OK: {"Hello from ESP32": "2"}
```

where i need use flush?
and make the mqtt size bigger is equal to make the client buffer sizer bigger? 

**Using**
C++2017 to compile.
Last versions of PubSubClient, SSLClient Linrarys

prototypicalpro, 2021-04-01:
Hello @HebertWP! I see a number of code errors in your sketch that is likely causing the issue. Could you try starting from the  [EthernetAWSIoT](https://github.com/OPEnSLab-OSU/SSLClient/tree/master/examples/EthernetAWSIoT) example (which I know works on the ESP32) and switching it to use WiFiClient? Afterwards I'd be happy to work with you if you still have the issue.

HebertWP, 2021-04-02:
i have re write the test from scratch, like you said and now it's works, i have forget to call setMutualAuthParams(), 12 hours to find this erro :( 
It's possible to add a Warning message abut this mistake, of forget to call setMutualAuthParams() before try connect with the client ? 
i'm very thankful to you

prototypicalpro, 2021-04-07:
Sorry to hear that! Unfortunately there isn't really a way for SSLClient to detect that it should be using mTLS other than calling `setMutualAuthParams` since the connection just times out if `setMutualAuthParams` is not called. I can add a note in the README to double check that this function is called, but that's probably the best I can do.

closed 2021-04-07 by prototypicalpro