OPEnSLab-OSU/SSLClient #39: AWS MQTT ESP32 SIM800 TinyGSM

aneggi, 2021-04-02:
Firts of all, thanks a lot to contribute. :-)

I am trining to connecting to AWS IOT by MQTT. I am using Esp32 with SIM800L. 
To manage SIM800L I am using TinyGSM library.
So I copied and paste your MQTT example, and tried with WIFI, and it works fine after some minor fix on the Ethernet library.
Now I added TinyGSM and removed WiFi but I got an error...

Partial important code changing here:
`
TinyGsm modem(SerialAT);
TinyGsmClient transport(modem);

SSLClient ethClientSSL(transport, TAs, (size_t)TAs_NUM, A5);
PubSubClient mqtt(mqttServer, 8883, callback, ethClientSSL);

`

I got this error:
`
Initializing modem...
[8304] ### TinyGSM Version: 0.10.9
[8304] ### TinyGSM Compiled Module:  TinyGsmClientSIM800
[8617] ### Modem: SIMCOM SIM800L
[8617] ### Modem: SIMCOM SIM800L
1
[NET] Signal quality before GPRS ( >10 OK ): 0
Modem Info: SIM800 R14.18
[29070] ### Network time zone updated.
[29073] ### Network time and time zone updated.
[29074] ### Daylight savings time state updated.
Waiting for network... success
Network connected
Attempting MQTT connection...(SSLClient)(SSL_ERROR)(m_update_engine): Error reading bytes from m_client. Write Error: 
(SSLClient)(SSL_ERROR)(m_update_engine): 0
(SSLClient)(SSL_ERROR)(connected): Not connected because write error is set
(SSLClient)(SSL_ERROR)(m_print_ssl_error): SSL_CLIENT_WRITE_FAIL
(SSLClient)(SSL_WARN)(m_run_until): Terminating with write error: 
(SSLClient)(SSL_WARN)(m_run_until): 4
(SSLClient)(SSL_ERROR)(m_start_ssl): Failed to initlalize the SSL layer
(SSLClient)(SSL_ERROR)(m_print_br_error): Unknown error code: 0
failed, rc=-2 try again in 5 seconds
(SSLClient)(SSL_ERROR)(connected): Not connected because write error is set
(SSLClient)(SSL_ERROR)(m_print_ssl_error): SSL_CLIENT_WRITE_FAIL
Attempting MQTT connection...(SSLClient)(SSL_ERROR)(connected): Not connected because write error is set
(SSLClient)(SSL_ERROR)(m_print_ssl_error): SSL_CLIENT_WRITE_FAIL
(SSLClient)(SSL_ERROR)(connected): Not connected because write error is set
(SSLClient)(SSL_ERROR)(m_print_ssl_error): SSL_CLIENT_WRITE_FAIL

`

Am I doing something bad when I use that "transport(modem)" in this way?
There is any way to find where is the error?

HebertWP, 2021-04-02:
i'm doing the same, now my code is in class so it' leatle diferente, but is essentially the same, here are some part of the code:
before void setup()
```
    serialAT_ = new HardwareSerial(1);
    modem_ = new TinyGsm(*serialAT_);
    clientnoSSL_ = new TinyGsmClient(*modem_);
    client_ = new SSLClient(*clientnoSSL_, TAs, (size_t)TAs_NUM, 27, 1, SSLClient::DebugLevel::SSL_WARN);
    mqtt_ = new PubSubClient(END_POINT_AWS, 8883, *client_);
```
in void setup()
```
    serialAT_->begin(9600, SERIAL_8N1, txPin_, rxPin_);
    mqtt_->setBufferSize(20000);
    client_->setMutualAuthParams(*mTLS);
    this->checkGPS_GPRS();
```
i have some problem with random internet lost, that i'm working on, now it's lost the connect with mqtt server some times, but with this code i'm able to reconnect automatic, by call this function in loop to  :
"Note: it's in first  version yet, so there is a loot of unnecessary print codes "
```
void checkGPS_GPRS()
{
    if (!modem_->getGPS(&lat, &lon))
        modem_->enableGPS();
    if (!modem_->isGprsConnected())
    {
        if (carrier_ == TIM)
        {
            modem_->gprsConnect(apnNameTim, apnUserTim, apnPswdTim);
            Serial.printf("TIM\r\n");
        }
        else if (carrier_ == VIVO)
            modem_->gprsConnect(apnNameVivo, apnUserVivo, apnPswdVivo);
        delay(1000);
    }else
        Serial.printf("Modem connected\r\n");

      if (!mqtt_->connected())
    {
        Serial.print("Attempting MQTT connection...\r\n");
        if (mqtt_->connect("ESP32-1"))
        {
            Serial.println("connected!!!!!!!!!!!!");
        }
        else
        {
            Serial.print("failed, rc=");
            Serial.println(mqtt_->state());
            client_->flush();
            delay(5000);
        }
    }
    else
    {
        Serial.printf("MQTT connected\r\n");
        if (client_->connected())
            while (client_->available() != 0)
                client_->read();
        mqtt_->loop();
        client_->flush();
    }
}
```
And i'm using SIM868

prototypicalpro, 2021-04-07:
Hello @aneggi! It looks like TinyGSM is failing to read from the modem, which might be intended behavior but I'm not sure. Could you [enable diagnostics in TinyGSM](https://github.com/vshymanskyy/TinyGSM#diagnostics-sketch) and post your logs with that?

HebertWP, 2021-04-12:
@aneggi In my case the library is working fine, can you test this code?, in your module, i mad it direct from example, i pretend test it with SIM7000G before try put this example direct in library, but it will take at lest a month to arrive from china in Brasil.

```
/*
  Connect to AWS IOT using SSLClient and Tinygsm Mdoule
   AWS_Root_CA.h is the trust anchor created using the Root CA from:
   https://www.amazontrust.com/repository/AmazonRootCA1.pem
   You can re-create it again using the python file present 
   in SSLClient/tools/pycert_bearssl/pycert_bearssl.py
   python pycert_bearssl.py convert --no-search <certificate PEM file>
   refer: https://github.com/OPEnSLab-OSU/SSLClient/issues/17#issuecomment-700143405
   
  Circuit:
   Ethernet shield WIZ850io:
   Power     2
   RX        16
   TX        4
   
  created 12 Abril 2021
  by Hebert Wandick Parreira, with EnthermetAWSIotExample as base, from Ram Rohit Gannavarapu
*/

// Select your modem:
// #define TINY_GSM_MODEM_SIM800
#define TINY_GSM_MODEM_SIM868

#include <Arduino.h>
#include <TinyGsmClient.h>
#include <SSLClient.h>
#include <PubSubClient.h>
#include "AWS_Root_CA.h" // This file is created using AmazonRootCA1.pem from https://www.amazontrust.com/repository/AmazonRootCA1.pem

#define THING_NAME "Teste"
#define MQTT_PACKET_SIZE 1024

#define APN_NAME "timbrasil.br"
#define APN_USER "tim"
#define APN_PSWD "tim"

void MQTTPublish(const char *topic, char *payload);
void updateThing();

static const char my_cert[] PROGMEM = R"KEY(
-----BEGIN CERTIFICATE-----
MIIDWjCCAkKgAwIBAgIVAMHgCQgsOQkBcsHxTbOkHUyHT2TYMA0GCSqGSIb3DQEB
CwUAME0xSzBJBgNVBAsMQkFtYXpvbiBXZWIgU2VydmljZXMgTz1BbWF6b24uY29t
IEluYy4gTD1TZWF0dGxlIFNUPVdhc2hpbmd0b24gQz1VUzAeFw0yMTA0MTAxMTQ1
MDlaFw00OTEyMzEyMzU5NTlaMB4xHDAaBgNVBAMME0FXUyBJb1QgQ2VydGlmaWNh
dGUwggEiMA0GCSqGSIb3EQEBAQUAA4IBDwAwggEKAoIBAQCzqzOhAnGbOPqQAAKs
URn/nx1W+uXErY/Pg/2Bpe2sWYoe+B0mzNpHNKKicF1ixgLqN1vdwl5vhelI9ypT
sJXTPmXK5DHSJ/l3YbBzZqGOeMjm8Rq01ilRPWQXJAvytz21n6utNj/Wr2j779RS
5ZsRKbCNd0zq1f33DJMABQLoWE+xw86UEvUP+PwRGCaUTqRxQKuTQNlbwMOOW5yg
I3dxAAfzQlgXUm4VcRMO/Q8z5NeNBDf0hqDiaq+Ye2OA1gAgnU37yHPqU5upw2ab
Ta+ImkkmDkirhU03LkyywxYWFPV+pi08kUvn+uKM7VgxMB0L+S8MmttJS9rVPK/k
UL2HAgMBAAGjYDBeMB8GA1UdIwQYMBaAFKrEmaBsLl1KrA2IBX9Um/xA1/opMB0G
A1UdDgQWBBSnb1bPri009jILq+Gck/80U6WUNDAMBgNVHRMBAf8EAjAAMA4GA1Ud
DwEB/wQEAwIHgDANBgkqhkiG9w0BAQsFAAOCAQEAfrvvVSHwJk9sbb3uPlbNwyd+
3+gw/BIghFPPjXjdfnIidLkBHMd+At2fkNH2bv9Xh4mEC7swrTUMaUC/hPXC2blw
vHEJUWsEfBXV36yZLHIz+mQwwNjRd/fyNgxVmakTGuEhU03Q6yAYCmZo5LzntzKM
3pGIv938V8U8Efr7GkJ3uBljd9IOiqt9aYE+XRAOq9kaZhhbzWVCuXMgdfn0Y6V1
Nl4Hes18FowwOgm2wz8ZrE0mAbjhGdjngTwkepfSI5ufGHDXiSCb7TMrzTOk9JwT
Dx/KNKVYjcCAnESukuOU+t0WGsWy99uog45E8qyZ/ZOrv/5PhaIhr6YdmWoTxA==
-----END CERTIFICATE-----
)KEY";

static const char my_key[] PROGMEM = R"KEY(
-----BEGIN RSA PRIVATE KEY-----
MIIEpQIBAAKCAQEAs6szoQJxmzj6kAACrFEZ/58dVvrlxK2Pz4P9gaXtrFmKHvgd
JszaRzSionBdYsYC6jdb3cJeb4XpSPcqU7CV0z5lyuQx0if5d2Gwc2ahjnjI5vEa
tNYpUT1kFyQL8rc9tZ+rrTY/1q9o++/UUuWbESmwjXdM6tX99wyTAAUC6FhPscPO
[... 122 lines cut ...]

void loop()
{
    if (!mqtt.connected())
    {
        reconnect();
    }
    mqtt.loop();
}

void updateThing()
{
    strcpy(publishPayload, "{\"state\": {\"reported\": {\"powerState\":\"ON\"}}}");
    MQTTPublish(publishShadowUpdate, publishPayload);
}

void MQTTPublish(const char *topic, char *payload)
{
    mqtt.publish(topic, payload);
    Serial.print("Published [");
    Serial.print(topic);
    Serial.print("] ");
    Serial.println(payload);
}
```

woodlist, 2021-04-16:
Trash away your SIM800L. It is not accommodated for SSL using. Just inquire your modem for SSL feature by AT command.
Instead, use SIM800C at least.

woodlist, 2021-04-16:
> Hello @aneggi! It looks like TinyGSM is failing to read from the modem, which might be intended behavior but I'm not sure. Could you [enable diagnostics in TinyGSM](https://github.com/vshymanskyy/TinyGSM#diagnostics-sketch) and post your logs with that?

I should report that the TinyGSM works perfect with ESP32 platform (namely Al-thinker ESP32-CAM and stand alone SIM800C)

woodlist, 2021-04-16:
Copy/Paste from my terminal:
`23:51:35.647 -> Waiting for network...
23:51:35.682 -> GPRS status: connected
23:51:35.682 -> Local IP:10.104.69.90
23:51:41.667 -> Connected to GPRS
23:51:41.667 -> 
23:51:41.667 -> Starting connection to server...
23:51:44.557 -> connected to server
23:51:45.781 -> HTTP/1.1 200 OK
23:51:45.781 -> Content-Type: text/html; charset=ISO-8859-1
23:51:45.781 -> Date: Fri, 16 Apr 2021 19:51:34 GMT
23:51:45.781 -> Expires: -1
23:51:45.781 -> Cache-Control: private, max-age=0
23:51:45.781 -> P3P: CP="This is not a P3P policy! See g.co/p3phelp for more info."
23:51:45.781 -> Server: gws
23:51:45.781 -> X-XSS-Protection: 0
23:51:45.781 -> X-Frame-Options: SAMEORIGIN
23:51:45.781 -> Set-Cookie: 1P_JAR=2021-04-16-19; expires=Sun, 16-May-2021 19:51:34 GMT; path=/; domain=.google.com; Secure
23:51:45.815 -> Set-Cookie: CGIC=; expires=Wed, 13-Oct-2021 19:51:34 GMT; path=/complete/search; domain=.google.com; HttpOnly
23:51:45.815 -> Set-Cookie: CGIC=; expires=Wed, 13-Oct-2021 19:51:34 GMT; path=/search; domain=.google.com; HttpOnly
23:51:45.815 -> Set-Cookie: NID=213=cv87kXlxnl_tO_ceOd121c3gaRX-Y1_-ji9k6sph52WQcEBqah4aoCZRJmpxfq4_iqO_ue4gBs6l6BLF_kTEZuVG_cUmQUf39M1IGO5WnDhJzuxcggTO3Yhvqtls4Bj15IU_5OkVAe0wPvE9udNlllrW0AzbrpWmDmNsrs_xnpA; expires=Sat, 16-Oct-2021 19:51:34 GMT; path=/; domain=.google.com; HttpOnly
23:51:45.849 -> Alt-Svc: h3-29=":443"; ma=2592000,h3-T051=":443"; ma=2592000,h3-Q050=":443"; ma=2592000,h3-Q046=":443"; ma=2592000,h3-Q043=":443"; ma=2592000,quic=":443"; ma=2592000; v="46,43"
23:51:45.849 -> Accept-Ranges: none
23:51:45.849 -> Vary: Accept-Encoding
23:51:45.849 -> Connection: close
23:51:45.849 -> Transfer-Encoding: chunked
23:51:45.849 -> 
23:51:45.849 -> 30ba
23:51:45.849 -> <!doctype html><html lang="hy"><head><meta charset="UTF-8"><meta content="/images/branding/googleg/1x/googleg_standard_color_128dp.png" itemprop="image"><title>arduino - Google &#1352;&#1408;&#1400;&#1398;&#1400;&#1410;&#1396;</title><script nonce="3giB/Ohtn4eWMUFxrZcyxg==">(function(){
23:51:46.189 -> document.documentElement.addEventListener("submit",function(b){var a;if(a=b.target){var c=a.getAttribute("data-submitfalse");a="1"==c||"q"==c&&!a.elements.q.value?!0:!1}else a=!1;a&&(b.preventDefault(),b.stopPropagation())},!0);document.documentElement.addEventListener("click",function(b){var a;a:{for(a=b.target;a&&a!=document.documentElement;a=a.parentElement)if("A"==a.tagName){a="1"==a.getAttribute("data-nohref");break a}a=!1}a&&b.preventDefault()},!0);}).call(this);(function(){
23:51:46.226 -> var a=window.performance;window.start=Date.now();a:{var b=window;if(a){var c=a.timing;if(c){var d=c.navigationStart,f=c.responseStart;if(f>d&&f<=window.start){window.start=f;b.wsrt=f-d;break a}}a.now&&(b.wsrt=Math.floor(a.now()))}}window.google=window.google||{};var h=function(g){g&&g.target.setAttribute("data-iml",Date.now())};document.documentElement.addEventListener("load",h,!0);google.rglh=function(){document.documentElement.removeEventListener("load",h,!0)};}).call(this);(function(){
23:51:46.260 -> var c=[],e=0;window.ping=function(b){-1==b.indexOf("&zx")&&(b+="&zx="+Date.now());var a=new Image,d=e++;c[d]=a;a.onerror=a.onload=a.onabort=function(){delete c[d]};a.src=b};}).call(this);</script><style>body{margin:0 auto;max-width:736px;padding:0 8px;}a{color:#1967D2;text-decoration:none;tap-highlight-color:rgba(0,0,0,.10)}a:visited{color:#4B11A8}a:hover{text-decoration:underline}img{border:0}html{font-family:Roboto,Helvetica Neue,Arial,sans-serif;font-size:14px;line-height:20px;text-size-adjust:100%;color:#3c4043;word-wrap:break-word;background-color:#fff}.bRsWnc{background-color:#fff;border-top:1px solid #dadce0;height:39px;overflow:hidden}.N6RWV{height:51px;overflow-scrolling:touch;overflow-x:auto;overflow-y:hidden}.Uv67qb{box-pack:justify;font-size:12px;line-height:37px;justify-content:space-between;justify-content:space-between}.Uv67qb a,.Uv67qb span{color:#70757a;display:block;flex:none;padding:0 16px;text-align:center;text-transform:uppercase;}span.OXXup{border-bottom:2px solid #4285f4;color:#4285f4;font-weight:bold}a.eZt8xd:visited{color:#70757a}.FElbsf{border-left:1px solid rgba(0,0,0,.12)}header article{overflow:visible}.Pg70bf{height:39px;display:box;display:flex;display:flex;width:100%}.H0PQec{position:relative;flex:1}.sbc{display:flex;width:100%}.Pg70bf input{margin:2px 4px 2px 8px;}.x{width:26px;color:#70757a;font:27px/38px arial, sans-serif;line-height:40px;}#qdClwb{flex:0 0 auto;width:39px;height:39px;border-bottom:0;padding:0;border-top-right-radius:8px;background-color:#1a73e8;border:1px solid #1558d6;background-image:url(data:image/gif;base64,R0lGODdhJAAjAPIHAODr/nCk+MPZ/FmV96zK+/7+/5K5+kqL9iwAAAAAJAAjAEADani63P4wykmbKcQRXDscQAEMXmmeaLQVLCukzyC09AjfeK7v/MAajACLhPMVAgwjsUcEiZa8xgAYrVqv2Kx2iwsIAAABknfBBAKrTE4IcMyot8ur8datqIbQfJdnAfo2WE6BV05wXIiJigkAOw==);}.sc{font-size:;position:absolute;top:39px;left:0;right:0;box-shadow:0px 2px 5px rgba(0,0,0,.2);z-index:2;background-color:#fff}.sc>div{padding:10px 10px;padding-left:16px;padding-left:14px;border-top:1px solid #dfe1e5}.scs{background-color:#f8f9fa;}.noHIxc{display:block;font-size:16px;padding:0 0 0 8px;flex:1;height:35px;outline:none;border:none;width:100%;-webkit-tap-highlight-color:rgba(0,0,0,.00);overflow:hidden;}.sbc input[type=text]{background:none}.sml .cOl4Id{display:none}.l{display:none}.sml header{background:none}.sml .l{display:block;padding:0 8px}.sml .l{letter-spacing:-1px;text-align:center;border-radius:2px 0 0 0; [... cut ...] 1xIPwAhUUcCsKHa4RBdwQzmd6BAgLEAw\x22\x3e\u0531\u0575\u056c \u0561\u0580\u0564\u0575\u0578\u0582\u0576\u0584\u0576\u0565\u0580\x3c/a\x3e\x3c/span\x3e\x3c/div\x3e\x3c/div\x3e\x3c/div\x3e\x3c/div\x3e\x3c/div\x3e');})();google.drty&&google.drty();</script></body></html>
23:52:25.707 -> 0
23:52:25.707 -> 
23:52:26.081 -> (SSLClient)(SSL_WARN)(connected): Socket was dropped unexpectedly (this can be an alternative to closing the connection)
23:52:26.081 -> 
23:52:26.081 -> disconnecting from server.
`

RupeshPilania, 2021-09-17:
@aneggi I have tested this library with some minor changes. Connection was stable till 5 hours at 2 messages per second publishing rate.
Observation: Frequent ssl errors were seen on one network(Vodafone) operator. SSL connection worked without any interruption for another network(Idea) for 5-6 hours. It might be due to network problems also. 
I had done few changes:
1. Changed buffer size from 2048 to the defined macro in a file(don't remember name, will update later) to get rid of ssl write warning.
2. Added flush after publishing/subscribing
Here is my code:
TinyGsm modem(serialAT);
TinyGsmClient ethClient(modem);
SSLClient ethClientSSL(ethClient, TAs, (size_t)TAs_NUM, 36,SSLClient::SSL_INFO);
void reconnect()
{
    while (!mqtt.connected()/*&& modem.isGprsConnected()*/)
    {
        Serial.print("Attempting MQTT connection...");
        if (mqtt.connect("arduinoClient"))
        {
            Serial.println("connected");
            for (int i = 0; i < 6; i++)
            {
                //        Serial.println(subscribeTopic[i]);
                mqtt.subscribe(subscribeTopic[i]);
                if (ethClientSSL.available()) { ethClientSSL.flush(); }
            }
            Serial.println("Started updateThing ");
            updateThing();
            Serial.println("Done updateThing ");
        }
        else
        {
            Serial.print("failed, rc=");
            Serial.print(mqtt.state());
            Serial.println(" try again in 5 seconds");
          //  if (ethClientSSL.available()) { ethClientSSL.flush(); }
            delay(5000);
        }
    }
}

void updateThing()
{
    strcpy(publishPayload, "{\"state\": {\"reported\": {\"powerState\":\"ON\"}}}");
    MQTTPublish(publishShadowUpdate, publishPayload);
}

void MQTTPublish(const char *topic, char *payload)
{
    
    mqtt.publish(topic, payload);
    if (ethClientSSL.available()) { ethClientSSL.flush(); }
    Serial.print("Published [ ");
    Serial.print(++i);
    Serial.print(" ");
    Serial.print(topic);
    Serial.print("] ");
    Serial.println(payload);
}

void gsm()
{
      static int uptime = 0;
      static int last,now;
      if (!modem.isGprsConnected()) {
      Serial.println("GPRS disconnected!");
      Serial.print(F("Connecting to "));
      Serial.print(APN_NAME);
      if (!modem.gprsConnect(APN_NAME, APN_USER, APN_PSWD)) {
        Serial.println(" fail");
        delay(10000);
        return;
      }
      if (modem.isGprsConnected()) { Serial.println("GPRS reconnected"); }
    }
    if (!mqtt.connected())
    {
        uptime = millis();
        i = 0;
        if (ethClientSSL.available()) { ethClientSSL.flush(); }
        reconnect();
    }
    mqtt.loop();    
    now = millis();
    if(now-last>10000){ 
    char msg[25] = {0};
    sprintf(msg,"uptime: %d", (millis()-uptime)/1000);     
    MQTTPublish("GsmClientConnected", msg);  
    last = now;  
    }
   
    
}

void gsm_setup()
{

    // Open serial communications and wait for port to open:
    serialAT.begin(115200, SERIAL_8N1, txPin, rxPin);
    Serial.begin(115200);
    pinMode(pwrPin, OUTPUT);
    pinMode(RELAY1, OUTPUT);
    pinMode(RELAY2, OUTPUT);
    digitalWrite(pwrPin, LOW);
     digitalWrite(RELAY1, LOW);
      digitalWrite(RELAY2, LOW);
    while (!Serial)
    {
        ; // wait for serial port to connect. Needed for native USB port only
    }

    ethClientSSL.setMutualAuthParams(mTLS);
    mqtt.setBufferSize(MQTT_PACKET_SIZE);

    // start the GSM connection:
    if (!modem.testAT())
    {
        Serial.printf("Turning on Modem\r\n");
        digitalWrite(pwrPin, HIGH);
        delay(3000);
        digitalWrite(pwrPin, LOW);
        delay(3000);
    }
    modem.restart();
    modem.gprsConnect(APN_NAME, APN_USER, APN_PSWD);
    if (modem.waitForNetwork())
        Serial.printf("NetWork Ok\r\n");
    else
        Serial.printf("NetWork Faill\r\n");
    // print your local IP address:
    Serial.print("My IP address: ");
    Serial.println(modem.localIP());
}

RupeshPilania, 2021-12-11:
Hi @adfo0 
My connection is very stable now. I had tested it for 24 hours with AWS.
Yes, I face these errors sometimes, probably due to poor network, but I am able to connect within few attempts.
Also please check if your gprs is connected before running mqtt loop every time.

biswaKL, 2022-03-07:
> Could you tell me if you have had similar problems and how I could face them. Thank you

Use these after Post:
```
        ethClientSSL.flush();
        ethClientSSL.clearWriteError();

```