OPEnSLab-OSU/SSLClient #41: It seems the issue at https://github.com/OPEnSLab-OSU/SSLClient/issues/27 need a resuscitation

woodlist, 2021-04-20:
**Describe the bug**
setVerificationTime gives compilation error. `error: 'setVerificationTime' was not declared in this scope
`
**To Reproduce**
Write some steps to reproduce the behavior. Be sure to include the want hardware required, if any.
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen. A reproducible sketch is especially helpful.

**Screenshots/Serial Output**
`error: 'setVerificationTime' was not declared in this scope
`

**Context (please complete the following information):**
 -  Al-thinker ESP32-CAM
 - Arduino Core Version 1.8.13

**Additional context**
I have used the algorithm introduced here; [https://www.geeksforgeeks.org/find-number-of-days-between-two-given-dates/](url) and successfully passed the performance. As arguments were used the result of code, introduced here [https://github.com/vshymanskyy/TinyGSM/blob/master/examples/AllFunctions/AllFunctions.ino](url)  given at 335-352 lines
Trying to put the difference between 1.1.1 and today days and elapsed seconds for the moment into "setVerificationTime" return compile error. What I do not do as required?
For more details, i should report that the whole code without given task doing well and interacts with telegram bot via GSM connection. Simply, I would like to eliminate possible certificate expiration in the future.

prototypicalpro, 2021-04-20:
Hello @woodlist! [`setVerificationTime`](https://openslab-osu.github.io/SSLClient/class_s_s_l_client.html#ab285c2f5a03124558ef7f74b9f3d12ad) is a member function on an SSLClient instance and not a standalone function, which means you'll need to call it like so:
```C++
SSLClient m_client(...);
...
m_client.setVerificationTime(<time>);
```
`setVerificationTime` was introduced in a relatively recent version of SSLClient, so I would also make sure the library is up to date.

In addition to the above, you'll also need to make sure that the `<time>` parameter for `setVerificationTime` is in the correct format, which is non-trivial as the time format is [a bit strange](https://bearssl.org/apidoc/bearssl__x509_8h.html#a7f3558b1999ce904084d578700b1002c). I haven't used TinyGSM before so I'm afraid I can't be of much help with the conversion, but if you have a [unix epoch](https://en.wikipedia.org/wiki/Unix_time) timestamp you can do the conversion as follows:
```C++
time_t timestamp = time(NULL);
m_client.setVerificationTime((uint32_t)(timestamp / 86400) + 719528), (uint32_t)(timestamp % 86400));
```
Hope this helps!

woodlist, 2021-04-20:
Oh, thanks for quick reply. Unfortunately, the TinyGSM does not UDP protocol backed, So epoch time thus is not available.
Please, provide a link for thanksgiving payment from an happy, as not possible to believe, user.

woodlist, 2021-04-20:
Please, advise about time zone. The mine is GMT +04:00. Do I need to deal with this or not?

prototypicalpro, 2021-04-20:
> Please, advise about time zone. The mine is GMT +04:00. Do I need to deal with this or not?

Yes, you should convert the time to UTC before passing it to `setVerificationTime`.

woodlist, 2021-04-21:
Thanks to @prototypicalpro the question has been solved.

closed 2021-04-21 by prototypicalpro