OPEnSLab-OSU/SSLClient #42: OTA integration

woodlist, 2021-04-21:
**Is your feature request related to a problem? Please describe.**
I did a try to use the SSLClient with OTA instrument, available here: [https://github.com/chrisjoyce911/esp32FOTA](url), but it was wholly based on Wifi and WifiSecure libs that is not the way of communication I need in.

**Describe the solution you'd like**
It would be perfect to get the OTA possibility added like introduced above library does, but having the end user free on physical communication device choice.

cotestatnt, 2021-04-22:
@woodlist  you could use the "core included" library [HTTPUpdate](https://github.com/espressif/arduino-esp32/tree/master/libraries/HTTPUpdate) which need to instantiated by passing the reference to the used client.

woodlist, 2021-04-22:
Please, help. Let we say, in OTA library is written `WiFiClientSecure clientForOta;
` How do it should be reformatted, having omitted the WiFiClientSecure, if I have the beneath client declared in main.ino?
`TinyGsmClient baseclient(modem);
SSLClient client(baseclient, TAs, (size_t)TAs_NUM, rand_pin);
`

HebertWP, 2021-04-23:
i looking inside the library, it's using setCACert() which is a WiFiClientSecure member( so no forced cast :/ ) , i see two ways to go:
   1- shadow WifiClientSecure class with a child and orverride using SSLClient-TinyGSM client hide inside, this way no work inside the  library is needed.
   2- fork the library and make the changes to use client interface direct.
 I'm going try to use the second approach, it's do't look hard, but it's going to take a time(principally because i had never set up a htpps OTA server properly), if i have success i will post in here

woodlist, 2021-04-23:
> i looking inside the library, it's using setCACert() which is a WiFiClientSecure member( so no forced cast :/ ) , i see two ways to go:
> 1- shadow WifiClientSecure class with a child and orverride [...]

I did a try to move all the OTA content from library to main code, for direct usage of defined by SSLClient client, but was unable to do advanced coding and make error free compilation

HebertWP, 2021-04-24:
i have find that espressif it's self have build a library caled Updade.h, this library just take a stream object(Arduino framework)  that have the firmaware in a binary format and write it in ship , to have write a very simple test.
```
#include <Arduino.h>
#include <FS.h>
#include <Update.h>
#include <SD.h>

void setup()
{
    Serial.begin(115200);
    SD.begin();
    Serial.printf("Updating...\r\n");
    if (SD.exists("/newFirmWare.bin"))
    {
        File file = SD.open("/newFirmWare.bin", FILE_READ);
        if (!Update.begin(file.size()))
        {
            Serial.printf("Updade is to big");
            delay(1000);
            ESP.restart();
        }
        Update.writeStream(file);
        while (!Update.end() && !Update.hasError())
            ;
        file.close();
        if (Update.hasError())
        {
            Update.printError(Serial);
            delay(1000);
            ESP.restart();
        };
        Serial.printf("Update done\r\n");
        delay(1000);
        ESP.restart();
    }
}

void loop()
{
}
```
the library have a example to [https://github.com/espressif/arduino-esp32/blob/master/libraries/Update/examples/SD_Update/SD_Update.ino](url)
So i wrote simple python prog to take my binary file break it in parts and send over mqqt protocol in RAW, but i'm having some problem with package lost .
In the library the are two more examples, AWS_S3_OTA_Update and HTTPS_OTA_Update, the fits one use wificlient direct.
Yet working, but it's look promising