OPEnSLab-OSU/SSLClient #43: SSLClient::connected "Socket was dropped unexpectedly" because m_client is done but SSLClient still has data in buffer

jandrassy, 2021-04-25:
**Describe the bug**
`SSLClient::connected` stops with "Socket was dropped unexpectedly" because m_client is done and returns false for `connected()` but SSLClient still has unread data in buffer. 

**To Reproduce**
I tested SSLClient library with my networking libraries EthernetENC and WiFiEspAT. Both end with this error in EthernetHTTPS.ino example. EthernetENC reads more data than WiFiEspAT but both stop reading before all data are read.
If I remove `if (!client.connected()) {` from the example all data are read.

**Expected behavior**
It is a valid state that m_client is already closed and doesn't have data so it returns false for `connected()`. So `SSLClient::connected` should not stop() and report disconnect while it still has data in buffer. `connected()` is specified to return true while data are available. https://www.arduino.cc/en/Reference/ClientConnected

**Screenshots/Serial Output**
```
Waiting for connection to WiFi
..
Connected to WiFi network.
connecting to www.arduino.cc...
Took: 4103
c_con: 1 br_con: 1 
(it repeats many times)
c_con: 1 br_con: 1
HTTP/1.1 200 OK
Date: Sun, 25 Apr 2021 14:05:03 GMT
Content-Type: text/plain
c_con: 0 br_con: 1
(SSLClient)(SSL_WARN)(connected): Socket was dropped unexpectedly (this can be an alternative to closing the connection)
c_con: 0 br_con: 0

disconnecting.
c_con: 0 br_con: 0
Received 80 bytes in 0.6543, rate = 0.12 kbytes/second
```

**Context:**
 - Arduino Core Version: Arduino SAMD core 1.8.9
 - Relevant Library Versions: WiFiEspAT 1.3.1 and EthernetENC 2.0.1 (GitHub master)
 - SSLClient Version 1.6.11

woodlist, 2021-04-29:
It seem, due to similar root my microcontroller connects to server 100% successful, but writing falls in 80% of cases.
Serial out is foolows:
`17:06:14.412 -> Connecting to server: vpn.bmwgate.tk
17:06:15.532 -> Connection successful!
17:06:17.572 -> (SSLClient)(SSL_ERROR)(m_update_engine): Error writing to m_client
17:06:17.572 -> (SSLClient)(SSL_ERROR)(m_update_engine): 0
17:06:17.572 -> (SSLClient)(SSL_ERROR)(connected): Not connected because write error is set
17:06:17.572 -> (SSLClient)(SSL_ERROR)(m_print_ssl_error): SSL_CLIENT_WRITE_FAIL
17:06:17.572 -> (SSLClient)(SSL_WARN)(m_run_until): Terminating with write error: 
17:06:17.572 -> (SSLClient)(SSL_WARN)(m_run_until): 4
17:06:17.572 -> (SSLClient)(SSL_ERROR)(write): Failed while waiting for the engine to enter BR_SSL_SENDAPP
17:06:17.612 -> (SSLClient)(SSL_ERROR)(write): Cannot operate if the write error is not reset: 
17:06:17.612 -> (SSLClient)(SSL_ERROR)(m_print_ssl_error): SSL_CLIENT_WRITE_FAIL
`

marcelstoer, 2022-04-08:
Could someone from the project team confirm this issue? I have the same stack as @jandrassy and I am seeing the same behavior.

prototypicalpro, 2022-04-08:
I haven't had this issue myself but the explanation makes sense. Unfortunately I don't have the bandwidth to try and implement a solution at the moment, but I'd be happy to review any PRs that people submit.

jandrassy, 2022-04-09:
> I haven't had this issue myself but the explanation makes sense. Unfortunately I don't have the bandwidth to try and implement a solution at the moment, but I'd be happy to review any PRs that people submit.

just add
```
if (available())
  return true;
```
at the beginning of `connected`

marcelstoer, 2022-04-09:
@jandrassy I generally trust your expertise a lot more than I trust mine wrt to C/C++/Arduino but your proposal doesn't fix the issue for me.

jandrassy, 2022-04-09:
> @jandrassy I generally trust your expertise a lot more than I trust mine wrt to C/C++/Arduino but your proposal doesn't fix the issue for me.

I see. Implementation of available() is wrong too

Decezaris, 2024-05-22:
> > @JAndrassy I generally trust your expertise a lot more than I trust mine wrt to C/C++/Arduino but your proposal doesn't fix the issue for me.
> 
> I see. Implementation of available() is wrong too

Same here... why is the `available()` method wrong?

Decezaris, 2024-05-22:
> @JAndrassy I generally trust your expertise a lot more than I trust mine wrt to C/C++/Arduino but your proposal doesn't fix the issue for me.

Any workaround about this issue? I tried @JAndrassy's reply but not worked here as well

Decezaris, 2024-05-22:
@JAndrassy thanks for the quick reply. I will try that here and report the results

Decezaris, 2024-05-22:
@JAndrassy I made both changes, but unfortunately same result :/

[image]

GM-90, 2025-02-17:
Hi, it's a pity that this error has not been fixed yet.
In the following code I leave this workaround that I made.
It will allow you to recover all the Client's data (with only one Warning).

`
bool Client_Connected=false ;
while(true){
        Ethernet.maintain();

        /*******************************************************/

        if( Client_Connected){
            //leer datos recibidos
            int len = client.available();
            if (len > 0) {
                byte buffer[80];
                if (len > 80){
                    len = 80;
                    client.read(buffer, len);
                    Serial.write(buffer, len);
                }
                else{
                    client.read(buffer, len);
                    Serial.write(buffer, len);
                    if( !client.connected() ){
                        Client_Connected=false ;
                        Serial.println("disconnecting.");
                        client.stop();
                    }
                }
      
            }
        }
        else{
            if ( client.connected() ) {
                Client_Connected=true;
            }
        }

        vTaskDelay(10);
    }
`