OPEnSLab-OSU/SSLClient #44: Error writing to m_client on ESP32

bkaraceylan, 2021-05-07:
**Describe the bug**
ESP32dev board with ec28j60 ethernet module.
Using UIPEthernet library.
Trust anchors are created using https://openslab-osu.github.io/bearssl-certificate-utility/ 
clt_ptr->connect(url.c_str(), 443) returns SSL Error. 

**To Reproduce**
Send a request to an SSL endpoint using the hardware setup mentioned above.

**Expected behavior**
Return html content.

**Screenshots/Serial Output**
(SSLClient)(SSL_ERROR)(m_update_engine): Error writing to m_client
(SSLClient)(SSL_ERROR)(m_update_engine): 0
(SSLClient)(SSL_ERROR)(connected): Not connected because write error is set
(SSLClient)(SSL_ERROR)(m_print_ssl_error): SSL_CLIENT_WRITE_FAIL
(SSLClient)(SSL_WARN)(m_run_until): Terminating with write error: 
(SSLClient)(SSL_WARN)(m_run_until): 4
(SSLClient)(SSL_ERROR)(m_start_ssl): Failed to initlalize the SSL layer
(SSLClient)(SSL_ERROR)(m_print_br_error): Unknown error code: 0

**Context (please complete the following information):**
platform = espressif32
board = esp32dev
framework = arduino
openslab-osu/SSLClient@^1.6.11
build_flags = 	-D BEARSSL_SSL_BASIC

bkaraceylan, 2021-05-08:
UIPEthernet seems to be the underlying cause. Closing this.

closed 2021-05-08 by bkaraceylan

woodlist, 2021-05-09:
This exact case happens with my connections very frequently too.

rohitsam, 2021-12-22:
Happens for me on esp32 wifi, any fix to this?

faizurrehman, 2022-05-04:
Is it fixed? I am facing similar issue with esp32

MeY0u, 2024-02-20:
still no fix?