OPEnSLab-OSU/SSLClient #45: Convert cert to Trust Anchor - empty certificates.h

GregorR1, 2021-05-18:
I'm attempting to use the pycert_bearssl.py script to convert the .crt file (PEM format) from https://test.mosquitto.org/

I followed the instructions at https://github.com/OPEnSLab-OSU/SSLClient/blob/master/TrustAnchors.md and I am able to run the script, but the output of certificates.h is empty.

I used the command:
py pycert_bearssl.py convert --no-search

As I said before, the script runs without issue, but the output file is empty.

I'm hoping it's something simple I've missed.

Thanks,
Gregor

prototypicalpro, 2021-05-19:
Hello! `pycert_bearssl` is a tool to convert the root certificate of a domain (otherwise known as the certificate authority) into a format accepted by bearssl. In order for it to do that, you must supply a root certificate to convert: since you did not supply any certificates, the tool runs but doesn't generate a useful output (as you said).

The root certificate for a given domain is typically either provided in the documentation or contained in a bundle such as [certifi](https://pypi.org/project/certifi/). In the case of `test.mosquito.com`, the root certificate is linked in the documentation:
> The encrypted ports support TLS v1.3, v1.2 or v1.1 with x509 certificates and require client support to connect. **For ports 8883 and 8884 you should use the certificate authority file ([mosquitto.org.crt (PEM format)](https://test.mosquitto.org/ssl/mosquitto.org.crt), or [mosquitto.org.der (DER format)](https://test.mosquitto.org/ssl/mosquitto.org.der)) to verify the server connection.** Port 8081 has a Lets Encrypt certificate, so you should use your system CA certificates or the appropriate Lets Encrypt CA certificate for verification.

Assuming you would like to connect to ports 8883 or 8884, all you should need to do to generate the trust anchors file is download `mosquitto.org.crt` and run:
```
py pycert_bearssl.py convert --no-search mosquitto.org.crt
```
Which will generate a header file containing data from `mosquitto.org.crt`.

roysG, 2022-01-02:
@prototypicalpro, In case i want to generate der for my own domain, for example mqtt-qa.some-domain.io

How should i need to find the crt file?

Today the ssl of the domain is hosted by cloudflare

prototypicalpro, 2022-01-03:
@roysG if your domain is hosted by cloudflare you can find their root certificate bundle [here](https://developers.cloudflare.com/ssl/origin-configuration/origin-ca#4-required-for-some-add-cloudflare-origin-ca-root-certificates). You can use this bundle with the `pycert_bearssl` tool to generate trust anchors.

closed 2022-01-03 by prototypicalpro