OPEnSLab-OSU/SSLClient #46: Discarded unread data to favor write operation

LenardValdez, 2021-05-26:
I'm using an Arduino Due with [Seeed Studio W5500 Ethernet Shield](https://www.seeedstudio.com/W5500-Ethernet-Shield.html) which has an Internal 32Kbytes Memory for TX/RX Buffers. 

My project involves only transmitting and receiving MQTT from a local broker, although I got to connect and transmit some MQTT messages I'm getting this error spammed on the serial and cannot receive any MQTT messages.

`(SSLClient)(SSL_WARN)(m_run_until): Discarded unread data to favor write operation`

[image]

at first I thought I messed up modifying the Ethernet Library from Arduino so I tried using EthernetLarge library fork provided by the guide but I'm still getting the same error and still cant receive MQTT

I just followed the provided example .ino file for MQTT with TLS, Is there something I missed doing to prevent this

prototypicalpro, 2021-05-26:
Yup, this is a known issue which I need to document (sorry). What's happening is SSLClient's internal buffer [`m_iobuf`](https://github.com/OPEnSLab-OSU/SSLClient/blob/1fe48948004a3ad52705ab26db4f65833ecb4e12/src/SSLClient.h#L469) is only large enough to allow communication in one direction at a time—because of this, if there is unread data in the buffer and the client attempts to write, SSLClient will discard the unread data instead of discarding the data that needs to be written. This works well for transactional connections like HTTP, but it causes problems with MQTT as there could be any number of messages going back and forth at a given time.

Unfortunately the only fix available at the moment is to modify the size of [`m_iobuf`](https://github.com/OPEnSLab-OSU/SSLClient/blob/1fe48948004a3ad52705ab26db4f65833ecb4e12/src/SSLClient.h#L469) in your copy of the library:
```C++
// replace
unsigned char m_iobuf[2048];
// with
unsigned char m_iobuf[BR_SSL_BUFSIZE_BIDI];
```
This modification isn't performed by default because on lower-end devices (Feather M0) there isn't enough memory available to have a buffer that large. The ESP32 has memory to spare, however, so there shouldn't be any issues.

LenardValdez, 2021-05-26:
Hello @prototypicalpro , thank you for responding.
will try this out

LenardValdez, 2021-05-27:
Thank you so much for the help @prototypicalpro, the warning is now gone.

closed 2021-05-27 by LenardValdez