OPEnSLab-OSU/SSLClient #47: error: 'br_x509_trust_anchor' does not name a type

jjauzion, 2021-06-18:
**Is your question related to a problem? Please describe.**
I'm trying to implement mutual TLS on Nano 33 IoT board for MQTT communication. I generated the `certificates.h` using the `pycert_bearssl.py` tool but I try to compile I got the following error:
```
certificates.h:60:14: error: 'br_x509_trust_anchor' does not name a type
 static const br_x509_trust_anchor TAs[] = {
              ^~~~~~~~~~~~~~~~~~~~
eggdome_sensor:40:30: error: 'TAs' was not declared in this scope
 SSLClient client(wifiClient, TAs, (size_t)TAs_NUM, A2);
                              ^~~
exit status 1
'br_x509_trust_anchor' does not name a type
```

Where is the `br_x509_trust_anchor` type supposed to be defined ? Is there a dependency I should include ?

**Additional context**
Here is a code generating this error:
```
#include "arduino_secrets.h"

#include "certificates.h"
#include <SSLClient.h>

#include <ArduinoMqttClient.h>
#include <WiFiNINA.h>

/////// please enter your sensitive data in the Secret tab/arduino_secrets.h
char ssid[] = SECRET_SSID;    // your network SSID (name)
char pass[] = SECRET_PASS;    // your network password (use for WPA, or use as key for WEP)

char mqttUser[] = SECRET_MQTT_USER;       // MQTT client username
char mqttPassword[] = SECRET_MQTT_PWD;    // MQTT client password

WiFiClient wifiClient;
SSLClient client(wifiClient, TAs, (size_t)TAs_NUM, A2);
MqttClient mqttClient(wifiClient);

const char  broker[] = "192.168.0.14";
int         port     = 8883;

void setup() {}

void loop() {}

```

prototypicalpro, 2021-06-18:
If you add `#include "SSLClient.h"` to the top of certificates.h that should fix it. I can also fix this issue with pycert_bearssl when I have time.

jjauzion, 2021-06-21:
Thanks, it works :)

closed 2021-06-23 by jjauzion