OPEnSLab-OSU/SSLClient #50: HandShake Sertificate

elexal, 2021-08-03:
hello friend
how can I use this function so that it recevies Auto handshake without certificate???
exp : when using class esp8266secure can get data handshake & its dosent need cert.h class or function & gets the certificate  automaticaly the same as chrome browser when searching the web site 
thx

prototypicalpro, 2021-08-05:
Hello! Unfortunately this functionality isn't implemented in SSLClient, as it's quite complicated and not as secure as using a header file. I don't have plans on adding this feature, but if you or anyone else would like to submit a PR I'd be happy to review it 🙂.

elexal, 2021-08-05:
hello prototypicalpro , thanks for your quick response
it would have been great if this feature existed 
i want to handshake the certificate like the browser , because my certificates (site , url) are changed every time .. :(