OPEnSLab-OSU/SSLClient #51: Hangs on `br_ssl_engine_recvrec_ack()` call

chudsaviet, 2021-08-05:
**Describe the bug**
I tried to use SSLClient on Arduino MKR NB 1500 with NB-IoT radio.
When I try to establish an HTTPS connection to `google.com`, it is getting stuck.
I have traced it down to `br_ssl_engine_recvrec_ack()` call, but I don't know how to add debug prints to BearSSL itself.

**To Reproduce**
I'm using Arduino MKR NB 1500 with PlatformIO.
1. Checkout project at https://github.com/buoys-dev/buoy/tree/experiment/sslclient , `experiment/sslclient` branch.
2. Open PlatformIO IDE, upload firmware to device.
3. Monitor serial port.

**Expected behavior**
Google's `robots.txt` displayed in the serial port.

**Context (please complete the following information):**
 - Arduino MKR NB 1500
 - PlatformIO
 - SSLClient latest from `master` branch - `1fe48948004a3ad52705ab26db4f65833ecb4e12`

**Screenshots/Serial Output**
Serial log:
<details>
  <summary>Click to expand!</summary>
```
Starting Arduino web client.
AT
OK
AT
OK
AT
OK
AT+CMEE=0
OK

+PACSP0
AT+CFUN=0
OK
AT+CPIN?
+CPIN: READY

OK
AT+CMGF=1
OK
AT+UDCONF=1,1
OK
AT+CTZU=1
OK
AT+CGDCONT=1,"IP",""
OK
AT+UAUTHREQ=1,0
OK
AT+CFUN=1
OK
AT+CEREG?
+CEREG: 0,0

OK
AT+CEREG?
+CEREG: 0,0

OK
AT+CEREG?
+CEREG: 0,0

OK
AT+CEREG?
+CEREG: 0,0

OK
AT+CEREG?
[... 854 lines cut ...]
OK
(SSLClient)(SSL_INFO)(m_run_until): Expected bytes count: 
(SSLClient)(SSL_INFO)(m_run_until): 3581
(SSLClient)(SSL_INFO)(m_update_engine): m_update_engine() BR_SSL_RECVREC section entered

+UUSORD: 0,1380
AT+USORD=0,512

+USORD: 0,512,"82072A2E79742E6265821A616E64726F69642E636C69656E74732E676F6F676C652E636F6D821B646576656C6F7065722E616E64726F69642E676F6F676C652E636E821C646576656C6F706572732E616E64726F69642E676F6F676C652E636E8218736F757263652E616E64726F69642E676F6F676C652E636E30210603551D20041A30183008060667810C010201300C060A2B06010401D679020503303C0603551D1F043530333031A02FA02D862B687474703A2F2F63726C732E706B692E676F6F672F6774733163332F517146786269394D3438632E63726C30820104060A2B06010401D6790204020481F50481F200F0007600EEC095EE8D72640F92E3C3B91BC712A3696A097B4B6A1A1438E647B2CBEDC5F90000017A9891D65D0000040300473045022100FF9843A5B759051498AFE16BBB14E6BE632BCBF72CDB8B93FC826AB81F009DAC022044D5D97205F87005F779883E821D7978A746F8C3AF22ED755B71BA9A397C29C90076007D3EF2F88FFF88556824C2C0CA9E5289792BC50E78097F2E6A9768997E22F0D70000017A9891D6860000040300473045022100A9F356998C8006ABA3DBDD6BC3A8F28445EF7F2958304637411F1513A8E3C9830220367E524513775637D19937A3C69492E1DDBA976F8C78861F95B75F348338C991300D06092A864886F70D01010B05000382010100C10B9E6B58EA5F31C8"
OK

+UUSORD: 0,868
(SSLClient)(SSL_INFO)(m_update_engine): m_update_engine() rlen > 0 section entered. Calling br_ssl_engine_recvrec_ack().
(SSLClient)(SSL_INFO)(m_update_engine): m_update_engine() br_ssl_engine_recvrec_ack() returned.
(SSLClient)(SSL_INFO)(m_update_engine): m_update_engine() BR_SSL_RECVREC section entered
AT+USORD=0,512

+USORD: 0,512,"251A49B6FCDDA646738414B12173BCBB5AA418ECA41A15B175D155431C45804C696F0F44D017A054885909EA8F9828BDE505C7586938864C7F53CC24225B5312F635607C2A6E66C2F55CF10C48BB6BD21240DDCF62FB9AA771C662B2877B13217774BBF04E2016288783B29E2B115301A775B2F61241193E989DF82F105DBD88EC00AA83F687E026CA6BCEA141A0BE21B4D1265CB536571C699A422BEF7591C7C04FEC0CAC0DE7C4ECEBE6B8065CAB396CF1EAEFA4E85F25D7278ADF1E62CA268A18AA38D472324AFD15F50561B8C541C16884D6B87EBD93B5F30AF32D1C81004F0369803F56676FFE5044401A641FDE4B8AA732BDE46900059A308205963082037EA003020102020D0203BC53596B34C718F5015066300D06092A864886F70D01010B05003047310B300906035504061302555331223020060355040A1319476F6F676C65205472757374205365727669636573204C4C43311430120603550403130B47545320526F6F74205231301E170D3230303831333030303034325A170D3237303933303030303034325A3046310B300906035504061302555331223020060355040A1319476F6F676C65205472757374205365727669636573204C4C43311330110603550403130A4754532043412031433330820122300D06092A864886F70D01010105000382010F003082010A0282010100F588DFE7628C1E37F8"
OK

+UUSORD: 0,356
(SSLClient)(SSL_INFO)(m_update_engine): m_update_engine() rlen > 0 section entered. Calling br_ssl_engine_recvrec_ack().

```
</details>

prototypicalpro, 2021-08-05:
Thanks for the detailed logs! The hanging is likely caused by the chip [hard faulting](https://www.embedded.com/debugging-hard-faults-in-arm-cortex-m0-based-socs/#:~:text=A%20hard%20fault%20is%20an,equal%20or%20higher%20than%20SVCall), which is typically caused by an invalid operation such as a null pointer dereference. Unfortunately according to your log this problem could be occurring anywhere in BearSSL, which is a rather large chunk of code. 

I wouldn't try print statement debugging BearSSL as it contains autogenerated code that will be difficult to work with. Since you're using the SAMD21, a better option is a tool called [FeatherTrace](https://github.com/OPEnSLab-OSU/FeatherTrace) which will insert itself in the hardfault handler and collect a stacktrace (full disclosure: I developed this tool). Another option is a debugger such as the [particle debugger](https://docs.particle.io/datasheets/accessories/debugger/), which works with [pio debug](https://docs.platformio.org/en/latest/plus/debugging.html) and will show you a call stack when the hard fault occurs. Either of these tools will give a line location of where the fault is occurring, which is (I think) the only way we could reasonably troubleshoot this further. Would those work for you?

prototypicalpro, 2021-08-05:
As a side note, it looks like the MKR libraries support TLS natively: [https://github.com/TelenorStartIoT/arduino-dev-kit-mqtt/blob/master/Arduino_M1_MQTT.ino](https://github.com/TelenorStartIoT/arduino-dev-kit-mqtt/blob/a8b5ac389e36abbc9bde920d61a796c9ec8c8422/Arduino_M1_MQTT.ino#L101-L111). Out of curiosity is there a reason you need SSLClient instead?

chudsaviet, 2021-08-18:
Now - actually no, I have found `ArduinoBearSSL` library, which works and uses hardware capabilities of Arduino MKR family. 
I have reported this bug just for it to be known.