OPEnSLab-OSU/SSLClient #52: Selfsigned certificate usage

woodlist, 2021-08-24:
**Is your question related to a problem? Please describe.**
To omit the dependency from trusted certificate issuers (theirs certificates are valid for short period of time) I did a decision to use own made self-signed certificate for my domain. This own certificate has 10 years long period of life. This is vital subject. Otherwise, I was forced to update the firmware of my controllers on certificate expiration, either, directly or by air, which is not convenient. Controllers in my project are tied with mqtt broker on my domain and are using secured mqtt protocol.
For now, the certificate generating engine at `https://openslab-osu.github.io/bearssl-certificate-utility/` does not return the code for www.bmwgate.tk
What to do to get this subject fixed, please advise.

prototypicalpro, 2021-09-02:
Hey @woodlist! You should be able to use the [pycert_bearssl](https://github.com/OPEnSLab-OSU/SSLClient/blob/master/tools/pycert_bearssl/pycert_bearssl.py) tool to generate the trust anchors instead:
```
python pycert_bearssl.py download www.bmwgate.tk
```
If that doesn't work (it should? but it may not) you can also manually [download the certificate yourself](https://medium.com/@menakajain/export-download-ssl-certificate-from-server-site-url-bcfc41ea46a2) and then convert the file using `pycert_bearssl`:
```
python pycert_bearssl.py --no-search <.crt or .pem file>
```

Hope this helps!

woodlist, 2021-10-29:
My experience over this issue gave an answer. The selfsigned certificate is useless. Free SSL certificate was not preferable for 3 months validity period and have bought 1 year long certificate at [https://www.onlydomains.com/](url)