OPEnSLab-OSU/SSLClient #54: ESP32 MQTT Communication with AWS works only one-way

barcesat, 2021-09-25:
Hello,
TL,DR: ESP32 with W5100 ethernet shield communication only works one way (sending, not receiving). might be due to the way the credentials are loaded?

I'm trying to migrate a code running on ESP32 that used to work with Wifi & WiFiClientSecure and has used the Arduino MQTT library - https://github.com/256dpi/arduino-mqtt.
The code has worked alright aside to disconnection issues of the WiFi that might be related to the physical site and thus we decided to test migration to an ethernet solution.
I plugged a W5100 ethernet shield, tested it successfully with the WebClient example and began migrating the whole product code to the ethernet solution based on EthernetAWSIoT.ino example.

The example works ok, however, I do not store the certificate and key as part of the code but read them to the memory using SPIFFS. this way I can keep them on the device and issue an update OTA with the credentials stored safely inside.

The problem I have is that this appears to be working only one way, that is the update of the MQTT topic goes as planned, but the subscribing or processing of the messages on a different topic simply does not happen.

I have also tested it to happen on the EthernetAWSIoT.ino example below.
I have also tried replacing the MQTT client to be 256dpi/arduino-mqtt instead of PubSubClient and that didn't work at all.
Any ideas where should I check for the lost data?

`#include <Arduino.h>
#include <SPI.h>
#include <EthernetLarge.h>
#include <SSLClient.h>
#include <PubSubClient.h>
#include "AWS_Root_CA.h"

//CERTS SPIFFS:
#include "secrets.h"
#include "FS.h"
#include "SPIFFS.h"

#define THING_NAME "******"
#define MQTT_PACKET_SIZE 1024
#define MQTT_KEEP_ALIVE 10
#define MQTT_TIMEOUT 5000

const char* mqttServer = "********.iot.us-east-1.amazonaws.com";
char publishPayload[MQTT_PACKET_SIZE];
char *subscribeTopic = "****/sub/" THING_NAME "";
char *publishTopic = "****/pub/" THING_NAME "";
void callback(char* topic, byte* payload, unsigned int length) 
{
  Serial.print("Message arrived [");
  Serial.print(topic);
  Serial.print("] ");
  for (int i=0;i<length;i++) 
  {
    Serial.print((char)payload[i]);
  }
  Serial.println();
  updateThing();
}

EthernetClient ethClient;
SSLClient ethClientSSL(ethClient, TAs, (size_t)TAs_NUM, 36);//, 1, SSLClient::SSL_INFO); // A0
PubSubClient mqtt_client(mqttServer, 8883, callback, ethClientSSL);

// Enter a MAC address for your controller below.
// Newer Ethernet shields have a MAC address printed on a sticker on the shield
byte mac[] = {
  0x00, 0xAA, 0xBB, 0xCC, 0xDE, 0x02
};

void reconnect() 
{
  while (!mqtt_client.connected()) 
  {
    Serial.print("Attempting MQTT connection...");
    if (mqtt_client.connect("arduinoClient")) 
    {
      Serial.println("connected");
    //   for (int i = 0; i < 2; i++) 
    //   {
    //    Serial.println(subscribeTopic[i]);
        // mqtt.subscribe(subscribeTopic[i]);
    //   }
        Serial.println(subscribeTopic);
        mqtt_client.subscribe(subscribeTopic);
        // ethClientSSL.flush();
        Serial.println("Started updateThing ");
        updateThing();
        Serial.println("Done updateThing ");

    } 
    else 
    {
      Serial.print("failed, rc=");
      Serial.print(mqtt_client.state());
      Serial.println(" try again in 5 seconds");
      delay(5000);
    }
  }
}

void setup() {
  // You can use Ethernet.init(pin) to configure the CS pin
  Ethernet.init(5);  // Most Arduino shields

  // Open serial communications and wait for port to open:
  Serial.begin(115200);
  while (!Serial) {
    ; // wait for serial port to connect. Needed for native USB port only
  }

  /*~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~*/
//CERTS SPIFFS:
  if(!SPIFFS.begin(FORMAT_SPIFFS_IF_FAILED)){
        Serial.println("SPIFFS Mount Failed");
        return;
    }
  // find certificate and thing files based on search strings
  findSigFiles(SPIFFS, "/", 0);

  // print file contents if available
  readFile(SPIFFS, AWS_CERT_CRT_FILEPATH);
  readFile(SPIFFS, AWS_CERT_PRIVATE_FILEPATH);

  // load file certs into arrays
  loadFile(AWS_CERT_CRT_FILEPATH, AWS_CERT_CRT_LOADED);
  loadFile(AWS_CERT_PRIVATE_FILEPATH, AWS_CERT_PRIVATE_LOADED);

  SPIFFS.end();

//end CERTS SPIFFS
/*~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~*/

  // start the Ethernet connection:
  Serial.println("Initialize Ethernet with DHCP:");
  if (Ethernet.begin(mac) == 0) {
    Serial.println("Failed to configure Ethernet using DHCP");
    if (Ethernet.hardwareStatus() == EthernetNoHardware) {
      Serial.println("Ethernet shield was not found.  Sorry, can't run without hardware. :(");
    } else if (Ethernet.linkStatus() == LinkOFF) {
      Serial.println("Ethernet cable is not connected.");
    }
    // no point in carrying on, so do nothing forevermore:
    while (true) {
      delay(1);
    }
  }
  // print your local IP address:
  Serial.print("My IP address: ");
  Serial.println(Ethernet.localIP());

   SSLClientParameters mTLS = SSLClientParameters::fromPEM(AWS_CERT_CRT_LOADED, 1220, AWS_CERT_PRIVATE_LOADED, 1679);
 ethClientSSL.setMutualAuthParams(mTLS);
  mqtt_client.setBufferSize(MQTT_PACKET_SIZE);
  mqtt_client.setKeepAlive(MQTT_KEEP_ALIVE);
  mqtt_client.setSocketTimeout(MQTT_TIMEOUT/1000);
}

void loop() {
  if (!mqtt_client.connected()) 
  {
    reconnect();
  }
  mqtt_client.loop();
}

void updateThing()
{
  strcpy(publishPayload, "{\"state\": {\"reported\": {\"powerState\":\"ON\"}}}");
//   MQTTPublish(publishShadowUpdate, publishPayload);
MQTTPublish(publishTopic, publishPayload);

}

void MQTTPublish(const char *topic, char *payload)
{
  mqtt_client.publish(topic, payload);
  Serial.print("Published [");
  Serial.print(topic);
  Serial.print("] ");
  Serial.println(payload);
}
`
Thanks!

prototypicalpro, 2021-09-25:
Hey @barcesat! In principle loading the credentials using SPIFFS should be fine so long as they remain in the same memory for the entire duration of the TLS connection, however I notice you've scoped the credentials object (`SSLClientParameters`) to `setup()`. This could be causing your issue: I wrote an [issue comment](https://github.com/OPEnSLab-OSU/SSLClient/issues/34#issuecomment-806081659) earlier that explains how to fix it:

> Hello @MaffooClock! Your second example doesn't work because SSLClient doesn't copy `mTLS`, which causes your code to use a dangling reference to the `mTLS` object on the stack and probably crash. This behavior is by design, as the `mTLS` object is very large and copying it would be too much for the lower-end microcontrollers SSLClient supports. This doesn't mean that changing the mutual authentication parameters dynamically is impossible, however—changing the certificates will work so long as the lifetime of the mutual authentication parameters is as long as the time they are active in SSLClient. In other words, `mTLS` should be somewhere that it won't get destroyed before SSLClient tries to reference it. One method of accomplishing this is using a global `unique_ptr`, which persists the parameters while still allowing them to be reallocated at runtime:
> 
> ```c++
> SSLClient sslClient( gsmClient, TAs, TAs_NUM, A6 );
> std::unique_ptr<SSLClientParameters> mTLS;
> 
> void setup()
> {
>   mTLS = std::unique_ptr<SSLClientParameters>(new SSLClientParameters(SSLClientParameters::fromPEM(...)));
>   sslClient.setMutualAuthParams(*mTLS);
>   // side note: the parameters to SSLClient::fromPEM are copied, so it is safe to destroy them afterwards
>   ...
> }
> 
> void main()
> {
>   // you can also change certificates on the fly with this approach:
>   if (...) {
>     Serial.println("Changing certificates!");
>     sslClient.stop();
>     mTLS = std::unique_ptr<SSLClientParameters>(new SSLClientParameters(SSLClientParameters::fromPEM(...)));
>     sslClient.setMutualAuthParams(*mTLS);
>     reconnect();
>     Serial.println("Changing certs completed!");
>   }
> }
> ```
> 
> There are many other ways you could go about this, but the above worked for me on my ESP32.

In addition, it's also possible that SSLClient is dropping packets due to it's unidirectional buffer. I wrote another [issue comment](https://github.com/OPEnSLab-OSU/SSLClient/issues/46#issuecomment-848908311) on this issue and how to fix it:

> Yup, this is a known issue which I need to document (sorry). What's happening is SSLClient's internal buffer [`m_iobuf`](https://github.com/OPEnSLab-OSU/SSLClient/blob/1fe48948004a3ad52705ab26db4f65833ecb4e12/src/SSLClient.h#L469) is only large enough to allow communication in one direction at a time—because of this, if there is unread data in the buffer and the client attempts to write, SSLClient will discard the unread data instead of discarding the data that needs to be written. This works well for transactional connections like HTTP, but it causes problems with MQTT as there could be any number of messages going back and forth at a given time.
> 
> Unfortunately the only fix available at the moment is to modify the size of [`m_iobuf`](https://github.com/OPEnSLab-OSU/SSLClient/blob/1fe48948004a3ad52705ab26db4f65833ecb4e12/src/SSLClient.h#L469) in your copy of the library:
> 
> ```c++
> // replace
> unsigned char m_iobuf[2048];
> // with
> unsigned char m_iobuf[BR_SSL_BUFSIZE_BIDI];
> ```
> 
> This modification isn't performed by default because on lower-end devices (Feather M0) there isn't enough memory available to have a buffer that large. The ESP32 has memory to spare, however, so there shouldn't be any issues.

Hope this helps!

barcesat, 2021-09-25:
@prototypicalpro Thank you so very very much!!
Before posting, I did try to read a lot of the issues that I thought would relate to that but haven't came across the ones you've mentioned.
I've applied both of the fixes and it worked :)
Then it led me to realize I had a bug in the callback function on the general code and I've fixed that as well :)

closed 2021-09-25 by barcesat

bvernham, 2022-08-29:
I tried the above with and MKR1010 and I got multiple error on compiling
'unique_ptr' in namespace 'std' does not name a template type

It highlighted the below line:

std::unique_ptr<SSLClientParameters> mTLS;

AWS_IoT_WiFi_SSLClient:48:6: error: 'unique_ptr' in namespace 'std' does not name a template type
 std::unique_ptr<SSLClientParameters> mTLS;
      ^~~~~~~~~~
C:\Users\bvernham\Dropbox (Isuzu North America)\Personal Backup\Arduino\MKR1010\AWS_IoT_WiFi_SSLClient\AWS_IoT_WiFi_SSLClient.ino: In function 'void setup()':
AWS_IoT_WiFi_SSLClient:56:3: error: 'mTLS' was not declared in this scope
   mTLS = std::unique_ptr<SSLClientParameters>(new SSLClientParameters(SSLClientParameters::fromPEM(SECRET_CERTIFICATE, sizeof SECRET_CERTIFICATE, SECRET_KEY, sizeof SECRET_KEY)));
   ^~~~
AWS_IoT_WiFi_SSLClient:56:15: error: 'unique_ptr' is not a member of 'std'
   mTLS = std::unique_ptr<SSLClientParameters>(new SSLClientParameters(SSLClientParameters::fromPEM(SECRET_CERTIFICATE, sizeof SECRET_CERTIFICATE, SECRET_KEY, sizeof SECRET_KEY)));
               ^~~~~~~~~~
AWS_IoT_WiFi_SSLClient:56:45: error: expected primary-expression before '>' token
   mTLS = std::unique_ptr<SSLClientParameters>(new SSLClientParameters(SSLClientParameters::fromPEM(SECRET_CERTIFICATE, sizeof SECRET_CERTIFICATE, SECRET_KEY, sizeof SECRET_KEY)));
                                          ^
'unique_ptr' in namespace 'std' does not name a template type