OPEnSLab-OSU/SSLClient #56: AWS IOT with SIMCOM7600SA

adfo0, 2021-10-14:
Hello everyone. I am working with an ESP32 board, with a SIMCOM7600SA modem.

I am using the SSL Client library to connect to AWS IOT Core. But I have some problems with the stability of the connection.

In some moments I get these errors, sometimes in the first connection attempt. or also after a few minutes.

After a few attempts to reconnect, it works again.

```
(SSLClient)(SSL_ERROR)(connect): Failed to connect using m_client. Are you connected to the internet?
(SSLClient)(SSL_ERROR)(connected): Not connected because write error is set
(SSLClient)(SSL_ERROR)(m_print_ssl_error): SSL_CLIENT_CONNECT_FAIL
[165043] ### Unhandled: ERROR
```

What could be the problem ? or is it normal?

Thanks

woodlist, 2021-10-29:
Are you sure that you have correctly set the data/time for SSL engine (sslClient.setVerificationTime)? 
I have an tested portion of code for this task. See beneath:
`// https://www.geeksforgeeks.org/find-number-of-days-between-two-given-dates/
using namespace std;
unsigned long Seconds;
// A date has day 'd', month 'm' and year 'y'
struct Date {
  int d, m, y;
};

// To store number of days in
// all months from January to Dec.
const int monthDays[12]
  = { 31, 28, 31, 30, 31, 30,
      31, 31, 30, 31, 30, 31
    };
int year3 = 0;
int month3 = 0;
int day3 = 0;
int hour3 = 0;
int min3 = 0;
int sec3 = 0;
float timezone = 0;

if (modem.getNetworkTime(&year3, &month3, &day3, &hour3, &min3, &sec3, &timezone)) {
calctime();
}

int countLeapYears(Date d)
{
  int years = d.y;

  // Check if the current year needs to be
  // considered for the count of leap years
  // or not
  if (d.m <= 2)
    years--;

  // An year is a leap year if it
  // is a multiple of 4,
  // multiple of 400 and not a
  // multiple of 100.
  return years / 4
         - years / 100
         + years / 400;
}

// This function returns number of
// days between two given dates
int getDifference(Date dt1, Date dt2)
{
  // COUNT TOTAL NUMBER OF DAYS
  // BEFORE FIRST DATE 'dt1'

  // initialize count using years and day
  long int n1 = dt1.y * 365 + dt1.d;

  // Add days for months in given date
  for (int i = 0; i < dt1.m - 1; i++)
    n1 += monthDays[i];

  // Since every leap year is of 366 days,
  // Add a day for every leap year
  n1 += countLeapYears(dt1);

  // SIMILARLY, COUNT TOTAL NUMBER OF
  // DAYS BEFORE 'dt2'

  long int n2 = dt2.y * 365 + dt2.d;
  for (int i = 0; i < dt2.m - 1; i++)
    n2 += monthDays[i];
  n2 += countLeapYears(dt2);

  // return difference between two counts
  return (n2 - n1);
}

int calctime()
{
  Date dt1 = { 1, 1, 0 };
  Date dt2 = { day3, month3, year3 };
  Seconds = hour3 * 3600 + min3 * 60 + sec3;
  sslClient.setVerificationTime((uint32_t)getDifference(dt1, dt2), (uint32_t)Seconds);
  SerialMon.println( "Difference between two dates is " + String(getDifference(dt1, dt2)) + " days");
  return 0;
}`

adfo0, 2021-11-01:
Hi @woodlist, this configuration should be done when the connection is for a long period of time? I did not know that it was necessary. Thanks for your answer

woodlist, 2021-11-09:
> Hi @woodlist, this configuration should be done when the connection is for a long period of time? I did not know that it was necessary. Thanks for your answer

How is your experience with suggested fix?

adfo0, 2022-03-09:
Finally, the problems were caused by the connectivity with the 3G network.

Apparently it is very sensitive to signal quality. When changing the antenna and looking for a sector with good signal quality, the problem did not appear again, before making any change.