OPEnSLab-OSU/SSLClient #58: Unstable connection via GSM network. The signal quality is strong enough.

woodlist, 2021-12-07:
**Describe the bug**
The code is fine tuned, everything works, but the performance of SSL engine is not stable. Particularly, when the internet traffic is absent, the SSL engine goes stuck. The attached video shows the local terminal and the Telegram bot performance. Even having terminal data showing that the connection is lost, Telegram bot commands are able to reach to controller.

**To Reproduce**
NONE

**Expected behavior**
A clear and concise description of what you expected to happen. A reproducible sketch is especially helpful.

**Screenshots/Serial Output**
[https://youtu.be/gj2La91Q32Q](url)
`16:42:58.120 -> Initializing modem...
16:43:04.492 -> Modem Info: SIM800 R14.18
16:43:04.492 -> Own number: Connecting to APN: internet.beeline.am OK
16:43:10.319 -> Waiting for network...
16:43:10.319 -> GPRS status: connected
16:43:10.319 -> Local IP:10.105.116.13
16:43:10.319 -> Requesting current network time
16:43:10.319 -> Year: 2021	Month: 12	Day: 7
16:43:10.319 -> Hour: 16	Minute: 43	Second: 9
16:43:10.319 -> Timezone: 4.00
16:43:10.319 -> Difference between two dates is 738495 days
16:43:21.490 -> got response
16:43:21.490 -> Message text: /start
16:43:29.926 -> got response
16:43:29.926 -> Message text: Engine
16:43:48.514 -> got response
16:43:48.514 -> Message text: Arm Car
16:43:53.305 -> MOTION DETECTED!!!
16:43:53.305 -> MOTION DETECTED!!!
16:43:53.305 -> MOTION DETECTED!!!
16:43:53.305 -> MOTION DETECTED!!!
16:43:53.305 -> MOTION DETECTED!!!
16:43:53.406 -> MOTION DETECTED!!!
16:43:53.406 -> MOTION DETECTED!!!
16:43:53.406 -> MOTION DETECTED!!!
16:43:53.406 -> MOTION DETECTED!!!
16:43:53.406 -> MOTION DETECTED!!!
16:43:53.406 -> MOTION DETECTED!!!
16:43:53.406 -> MOTION DETECTED!!!
16:43:53.440 -> MOTION DETECTED!!!
16:43:53.547 -> MOTION DETECTED!!!
16:43:53.580 -> MOTION DETECTED!!!
16:43:53.614 -> MOTION DETECTED!!!
16:43:53.614 -> MOTION DETECTED!!!
16:43:53.614 -> MOTION DETECTED!!!
16:43:53.614 -> MOTION DETECTED!!!
16:43:53.614 -> MOTION DETECTED!!!
16:43:53.614 -> MOTION DETECTED!!!
16:43:53.614 -> MOTION DETECTED!!!
16:43:53.614 -> MOTION DETECTED!!!
16:43:53.614 -> MOTION DETECTED!!!
16:43:53.614 -> MOTION DETECTED!!!
16:43:53.614 -> MOTION DETECTED!!!
16:43:53.614 -> MOTION DETECTED!!!
16:43:53.614 -> MOTION DETECTED!!!
16:43:53.614 -> MOTION DETECTED!!!
16:43:53.614 -> MOTION DETECTED!!!
16:43:53.614 -> MOTION DETECTED!!!
16:43:53.614 -> MOTION DETECTED!!!
16:43:53.614 -> MOTION DETECTED!!!
16:43:53.614 -> MOTION DETECTED!!!
16:43:53.648 -> MOTION DETECTED!!!
16:43:53.648 -> MOTION DETECTED!!!
16:43:53.648 -> MOTION DETECTED!!!
16:43:53.648 -> MOTION DETECTED!!!
16:43:53.648 -> MOTION DETECTED!!!
16:43:53.648 -> MOTION DETECTED!!!
16:43:53.648 -> MOTION DETECTED!!!
16:43:53.648 -> MOTION DETECTED!!!
[... 189 lines cut ...]
16:56:24.859 -> (SSLClient)(SSL_ERROR)(m_update_engine): 0
16:56:24.859 -> (SSLClient)(SSL_ERROR)(connected): Not connected because write error is set
16:56:24.893 -> (SSLClient)(SSL_ERROR)(m_print_ssl_error): SSL_CLIENT_WRITE_FAIL
16:56:25.334 -> (SSLClient)(SSL_WARN)(m_run_until): Terminating with write error: 
16:56:25.368 -> (SSLClient)(SSL_WARN)(m_run_until): 4
16:56:25.368 -> (SSLClient)(SSL_ERROR)(m_start_ssl): Failed to initlalize the SSL layer
16:56:25.368 -> (SSLClient)(SSL_ERROR)(m_print_br_error): Unknown error code: 0
16:56:25.368 -> (SSLClient)(SSL_ERROR)(connected): Not connected because write error is set
16:56:25.368 -> (SSLClient)(SSL_ERROR)(m_print_ssl_error): SSL_CLIENT_WRITE_FAIL
16:56:25.402 -> (SSLClient)(SSL_ERROR)(connected): Not connected because write error is set
16:56:25.402 -> (SSLClient)(SSL_ERROR)(m_print_ssl_error): SSL_CLIENT_WRITE_FAIL
16:56:26.421 -> (SSLClient)(SSL_ERROR)(connected): Not connected because write error is set
16:56:26.421 -> (SSLClient)(SSL_ERROR)(m_print_ssl_error): SSL_CLIENT_WRITE_FAIL
16:56:47.282 -> (SSLClient)(SSL_WARN)(m_run_until): Terminating because the ssl engine closed
16:56:47.316 -> (SSLClient)(SSL_ERROR)(m_start_ssl): Failed to initlalize the SSL layer
16:56:47.316 -> (SSLClient)(SSL_ERROR)(m_print_br_error): Unknown error code: 307`

**Context (please complete the following information):**
 - Device Type: ESP32
 - Arduino Core Version: 1.8.16
 - Relevant Library Versions [PubSubClient, etc.]
 - SSLClient Version: Up-to-date

**Additional context**
Add any other context about the problem here.

HebertWP, 2021-12-07:
i believe that this type of problem is due the low memory stack in sim800,  i have put a sim7000g working in same code, that i have been using in sim868 with high data size package to AWS, 4G mode it had not encounter this type of problem not a single time in 1 hour of operation.

woodlist, 2021-12-26:
Upon newest Arduino IDE upgrade the compilation gives
`C:\Users\user\Documents\Arduino\libraries\SSLClient\src\SSLClientParameters.cpp: In function 'void std::__throw_length_error(const char*)':
C:\Users\user\Documents\Arduino\libraries\SSLClient\src\SSLClientParameters.cpp:5:67: warning: 'noreturn' function does return
     void __attribute__((weak)) __throw_length_error(char const*) {}`
Compared with earlier Arduino 8.16 IDE, global variables used memory has been dropped down dramatically. Seems IDE developers have too much hard work done.

prototypicalpro, 2022-01-03:
Yup those warnings are fine, they are an artifact of a hack that allows vectors to be used on older versions of the arduino core.