OPEnSLab-OSU/SSLClient #59: Can't get AWSIoT example working on ESP32

liamcollins, 2021-12-23:
I'm almost certainly doing something dumb, but I've been at it for a while and I can't figure out what my problem is.

I'm using the AWSIoT example, substituting wifi for ethernet in order to use an ESP32 without an ethernet shield. I've included EthernetLarge.h. I think I generated my trust anchors correctly using pycert_bearssl.py. I know that the certs work because I can connect using wificlientsecure just fine. However I need this for another board (automation direct P1AM) which does not have Wifi so I need to figure out how to get the connection working without using wificlientsecure.

Any help greatly appreciated!

Code:
```
#include <SPI.h>
#include <EthernetLarge.h>
#include <SSLClient.h>
#include <PubSubClient.h>
#include "certificates.h" // This file is created using AmazonRootCA1.pem from https://www.amazontrust.com/repository/AmazonRootCA1.pem

// Wifi Info
// ***********************************************************
const char* ssid       = "...";
const char* password   = "...";

#define THING_NAME "..."
#define MQTT_PACKET_SIZE  1024

const char my_cert[] = \
"-----BEGIN CERTIFICATE-----\n" \ 
"MIIDWTCCAkGgAwIBAgIUCfyz78MYjtdWqZzv/uYK49pfIvYwDQYJKoZIhvcNAQEL\n" \
...
"-----END CERTIFICATE-----\n";

const char my_key[] = \
"-----BEGIN RSA PRIVATE KEY-----\n" \ 
"MIIEowIBAAKCAQEAoy+7EXAGAnSekgHJuo9eWzL8/Le1KNSxbhY5BtiOIAT+8FdC\n" \
...
"IVwcDl+IbWqy/Eg1Ux1olMYXoG1+iB5VRK2MpBiwoOrvvRE0Q3nD" \
"-----END RSA PRIVATE KEY-----\n";

SSLClientParameters mTLS = SSLClientParameters::fromPEM(my_cert, sizeof my_cert, my_key, sizeof my_key);

const char* mqttServer = "xxxxxxxxxx-ats.iot.us-east-2.amazonaws.com";

void callback(char* topic, byte* payload, unsigned int length) 
{
  Serial.print("Message arrived [");
  Serial.print(topic);
  Serial.print("] ");
  for (int i=0;i<length;i++) 
  {
    Serial.print((char)payload[i]);
  }
  Serial.println();
}

WiFiClient ethClient;
SSLClient ethClientSSL(ethClient, TAs, (size_t)TAs_NUM, 5, 1, SSLClient::SSL_INFO);
PubSubClient mqtt(mqttServer, 8883, callback, ethClientSSL);

// Enter a MAC address for your controller below.
// Newer Ethernet shields have a MAC address printed on a sticker on the shield
byte mac[] = {
  0xDE, 0xAD, 0xBE, 0xEF, 0xFE, 0xED
};

void reconnect() 
{
  while (!mqtt.connected()) 
  {
    Serial.print("Attempting MQTT connection...");
    if (mqtt.connect("arduinoClient")) 
    {
      Serial.println("connected");
    } 
    else 
    {
      Serial.print("failed, rc=");
      Serial.print(mqtt.state());
      Serial.println(" try again in 5 seconds");
      delay(5000);
    }
  }
}

void connectToWifi() {
  // Connect to Wi-Fi
  Serial.println("Connecting to Wifi");
  
  WiFi.begin(ssid, password);
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }
  Serial.println("WiFi connected.");
}

void setup() {
  // Open serial communications and wait for port to open:
  Serial.begin(115200);
  while (!Serial) {
    ; // wait for serial port to connect. Needed for native USB port only
  }
  
  connectToWifi();

  ethClientSSL.setMutualAuthParams(mTLS);
  mqtt.setBufferSize(MQTT_PACKET_SIZE);
  
}

void loop() {
  if (!mqtt.connected()) 
  {
    reconnect();
  }
  mqtt.loop();
}
```

Serial output:
```
...WiFi connected.
Attempting MQTT connection...(SSLClient)(SSL_INFO)(connect): Base client connected!
(SSLClient)(SSL_INFO)(m_start_ssl): about to inject random data from pin
(SSLClient)(SSL_INFO)(m_start_ssl): entropy injected!
(SSLClient)(SSL_INFO)(m_print_br_state): (SSLClient)(SSL_INFO)(m_print_br_state): State: 
   SENDREC
(SSLClient)(SSL_INFO)(m_print_br_state): (SSLClient)(SSL_INFO)(m_print_br_state): State: 
   RECVREC
(SSLClient)(SSL_INFO)(m_run_until): m_run changed state:
(SSLClient)(SSL_INFO)(m_print_br_state): (SSLClient)(SSL_INFO)(m_print_br_state): State: 
   RECVREC
(SSLClient)(SSL_INFO)(m_run_until): Expected bytes count: 
(SSLClient)(SSL_INFO)(m_run_until): 5
(SSLClient)(SSL_INFO)(m_print_br_state): (SSLClient)(SSL_INFO)(m_print_br_state): State: 
   SENDREC
(SSLClient)(SSL_INFO)(m_print_br_state): (SSLClient)(SSL_INFO)(m_print_br_state): State: 
   RECVREC
(SSLClient)(SSL_ERROR)(m_run_until): SSL internals timed out! This could be an internal error, bad data sent from the server, or data being discarded due to a buffer overflow. If you are using Ethernet, did you modify the library properly (see README)?
(SSLClient)(SSL_ERROR)(connected): Not connected because write error is set
(SSLClient)(SSL_ERROR)(m_print_ssl_error): SSL_BR_WRITE_ERROR
(SSLClient)(SSL_ERROR)(m_start_ssl): Failed to initlalize the SSL layer
(SSLClient)(SSL_ERROR)(m_print_br_error): Unknown error code: 0
failed, rc=-2 try again in 5 seconds
```

prototypicalpro, 2021-12-24:
Your code looks fine, and I've never heard of a timeout on the ESP32. A common issue SSLClient and MQTT is the internal buffer which is rather small by default: try applying the fix in [this issue comment](https://github.com/OPEnSLab-OSU/SSLClient/issues/46#issuecomment-848908311) and see if that changes anything.

liamcollins, 2021-12-24:
Thanks @prototypicalpro for the speedy reply!

I made the change you suggest to the internal buffer. Unfortunately still the same output.

liamcollins, 2021-12-25:
Got it!
Stupid mistake on my part.

I changed the certs to the following format and it worked:
```
static const char my_cert[] PROGMEM = R"KEY(
-----BEGIN CERTIFICATE-----
MIIDRTCCAkGgAwIBAgIUCfyz78MYjtdWqZzv/uYK49pfIvYwDQYJKoZIhvcNAQEL
...
-----END CERTIFICATE-----

)KEY";
```

So either that slash at the end of each line of the cert wasn't working as I was expecting or that's a convention I didn't realize actually means that all that stuff should be on one line in the code, or I had some other silly little copy/paste or formatting error.

I didn't look too hard at the certs because I was thinking I'd get some error pointing me in that direction if that was the problem rather than a timeout. But the more you know!

Thanks for the help, and Merry Christmas all! 

Closing the issue.

closed 2021-12-25 by liamcollins