OPEnSLab-OSU/SSLClient #61: Disable trust anchor

roysG, 2022-01-02:
As i understand the trust anchor working according my domain certificate.
As i understand, In case the domain ssl is expire, than the trust anchor also should be updated.

In my case i do not have the option to updating the trust anchor. 
How can i disable it?

prototypicalpro, 2022-01-03:
This is a duplicate of #22. I don't have the bandwidth to implement the fix proposed in that issue, but I'd be happy to review a PR on it.

closed 2022-01-03 by prototypicalpro

roysG, 2022-01-03:
Do i need to uodate the trust anchor manually, in case the ssl certificate
is upated after a year?

marcelstoer, 2022-04-08:
Depends on what you mean by "manually". For devices out in the field you would normally implement some form of OTA (over the air update). You will then be able to have the device update itself i.e. it'll fetch the new firmware that contains the updated trust anchors.