OPEnSLab-OSU/SSLClient #68: Cannot operate on a closed SSL connection

radu-bethub, 2022-03-29:
I'm getting 

13:17:55.152 > (SSLClient)(SSL_ERROR)(available): Cannot operate on a closed SSL connection.
13:17:55.152 > (SSLClient)(SSL_ERROR)(m_print_br_error): Incoming record is too large to be processed, or buffer is too small for the handshake message to send.

while trying to get my firmware bin file.

EthernetClient ClientESP;
SSLClient clientESP(ClientESP, TAs, (size_t)TAs_NUM, 25);
HttpClient clientHttp = HttpClient(clientESP, server, 443);

any idea? my GET server address is hosted on cloudflare.

mikeme2009, 2022-03-29:
Thank you. I will try

radu-bethub, 2022-03-29:
well done @prototypicalpro. I managed to increase the buffer directly from library for testing and now the update OTA is working well. I will continue to work in order to don't alter the library directly and increase that variable from my env. thanks a lot for you help.

javierdelavega, 2024-10-18:
Hello. I have ran in the same issue (only when using Cloudflare) and your solution worked for me, so thanks @prototypicalpro.
@radu-bethub did you managed to increase the buffer size with an env setting? I can't figure out how to do that, since the buffer size seems to be hardcoded in the SSLCLient.h. Am I missing something?