OPEnSLab-OSU/SSLClient #82: Mutual TLS Authentification - Read Certificates from a Filesystem

SJDickinsonSDS, 2022-11-21:
Hi, I'm using your library to connect to AWS's IoT Core using MQTT over TLS. 

I'd like to be able to store my certificates in a filesystem, namely SPIFFS and read them from there before passing them to SSLParameters::. My reading of the documentation and source code suggests that isn't really possible as SSLParameters:: needs to be called before a filesystem would normally be mounted in setup(). It seems this would limit the library to hard-coded certificates and thus make certificate exchange difficult?

I could be missing something, is there a way to store certificate in a filesystem and use them with this library?

bakadave, 2023-02-28:
I have the same question.

scsPEp, 2023-03-13:
I think following issue already has the solution for your problems. At least it worked for me.

https://github.com/OPEnSLab-OSU/SSLClient/issues/34