OPEnSLab-OSU/SSLClient #83: No such file or directory  #include <vector>

ggerminiani, 2022-12-08:
Hey everybody... I'm just trying to run the example, but I get this message: 

> In file included from /Users/XXX/Documents/Arduino/libraries/SSLClient-1.6.11/src/SSLClient.h:23:0,
>                  from /Users/XXX/Documents/Arduino/libraries/SSLClient-1.6.11/examples/EthernetMQTT/EthernetMQTT.ino:19:
> /Users/XXX/Documents/Arduino/libraries/SSLClient-1.6.11/src/SSLClientParameters.h:31:10: fatal error: vector: No such file or directory
>  #include <vector>

Some advice?

Thanks a lot

djaus2, 2023-03-12:
Me tooin this issue.
```
In file included from c:\Users\david\Documents\Arduino\libraries\SSLClient\src/SSLClient.h:23:0,
                 from C:\Users\david\repos\ArduinoSSL\SSLClient\examples\EthernetMQTT\EthernetMQTT.ino:22:
c:\Users\david\Documents\Arduino\libraries\SSLClient\src/SSLClientParameters.h:31:10: fatal error: vector: No such file or directory
 #include <vector>
          ^~~~~~~~
compilation terminated.
```

djaus2, 2023-03-12:
I think there is a need for additional C++ libraries such as cstdlib, vector etc.???
These are included with my RPi Pico W BSP.
I am trying to use Arduino Uno.