OPEnSLab-OSU/SSLClient #90: how to set the root certificate in pem format as a trust anchor

mayankbytebeam, 2023-03-02:
I have the root certificate to verify the server identity in the below format, how to set this out.

const char root_ca[] = 
"-----BEGIN CERTIFICATE-----\n"
"MIIDpDCCAowCCQC7mCk5Iu3YmDANBgkqhkiG9w0BAQUFADCBkzELMAkGA1UEBhMC\n"
...
"-----END CERTIFICATE-----\n";

I checked out the example codes but couldn't find the solution.

andersruneson, 2023-05-03:
I'm not sure I understand the question.

But you could use the tool below to generate the contents, that is the easiest way.
https://openslab-osu.github.io/bearssl-certificate-utility/

Then paste the output to the trust_anchors.h file in your project.

If you have a root cert in PEM format then you can use the tool below.
https://github.com/OPEnSLab-OSU/SSLClient/tree/master/tools/pycert_bearssl

> $ python3 pycert_bearssl.py convert your_cert.pem

mayankbytebeam, 2023-05-03:
@andersruneson Thanks for coming up with this.

So my requirement is like root certificate will be stored in the file system partition (say spiffs) along with other broker information and during initialization phase I will read up the file system and connect to the broker. I can't change the format to be writing into file system in this case.

Please see our official [sdk](https://github.com/bytebeamio/bytebeam-arduino-sdk) and let us know if there is any possibility for us to use the SSLClient library.

andersruneson, 2023-05-03:
Ah, so want to convert a PEM-formatted root certificate on the device itself. That is not impossible, but you have to write some code yourself that converts a PEM-formatted certificate to the format used by bearssl, similar to the pycert_bearssl.py-tool. Unless you find a tool that does exactly this.

mayankbytebeam, 2023-05-04:
can you add this as a feature for the library to be able to use certificate in the PEM format?

andersruneson, 2023-05-05:
Well, I could probably do it if you pay me.
I came across this library which seems to have the cert in pem format:

cmdgeofease, 2024-01-30:
> I'm not sure I understand the question.
>
> But you could use the tool below to generate the contents, that is the easiest way. https://openslab-osu.github.io/bearssl-certificate-utility/
>
> Then paste [...]

Hello! For this tool you have above, do you know the expiry dates or duration of the certificates that it generates?

Thanks!

andersruneson, 2024-01-31:
I think the tools just converts a cert to bear format, so the expiry is set when you generate the cert in the first place.

cmdgeofease, 2024-01-31:
> I think the tools just converts a cert to bear format, so the expiry is set when you generate the cert in the first place.

Thank you for the quick response. How about for the following tool? https://openslab-osu.github.io/bearssl-certificate-utility/ 

I created a trust anchor but am unsure the expiry date of it based on this tool.

thanks!

aremon78, 2024-06-08:
thank you for this discussion, will try to convert ca to trust_anchor. do any of you know free mqtt broker that could generate ca, client key and client cert?

aremon78, 2024-06-08:
https://github.com/OPEnSLab-OSU/SSLClient/tree/master/tools/pycert_bearssl 
try use this but always return Could not find a root certificate for certificate.crt
Wrote 0 trust anchors to certificates.h
any idea why?